GPSBabel/gpsbabel #448: Use the track filter as suggested above.  If you are using the command line 1.6.0 should work using a shift by 7168d(ays).  But if you are using the GUI you need a fix from after 1.6.0,

MAlfare, 2019-12-29:
Use the track filter as suggested above.  If you are using the command line 1.6.0 should work using a shift by 7168d(ays).  But if you are using the GUI you need a fix from after 1.6.0,
https://github.com/gpsbabel/gpsbabel/pull/363.  There is a pre-release version that includes this fix at https://github.com/gpsbabel/gpsbabel/releases, either continuous for macos or continuous-windows for windows.  The fix also adds the ability to shift by 1024w(eeks) with the command line.

_Originally posted by @tsteven4 in https://github.com/gpsbabel/gpsbabel/issues/349#issuecomment-526851146_

MAlfare, 2019-12-29:
As gpsbabel reads tracks from mentioned device but with no waypoints,  I use mtkbabel to read from device. I can convert the Holux  .bin-file I get to .gpx with gpsbabel filter: "-x track,move=7169d0h0m0s" but time/date of waypoints is unchanged. 
Is there a way to change them too?

MAlfare, 2019-12-29:
Tried to change the title in error into: "Holux m241, waypoints, track filter move" but am not able to :-(

tsteven4, 2019-12-29:
It's a bit ugly, but you could use a combination of the nuketypes, transform and track filters.
nuke the tracks, transform wpts to trks, run track,move, transform trks back to wpts.

MAlfare, 2019-12-29:
> It's a bit ugly, but you could use a combination of the nuketypes, transform and track filters.
> nuke the tracks, transform wpts to trks, run track,move, transform trks back to wpts.
How could I do that? Have a gpx-file containing track(s) and waypoints.

tsteven4, 2019-12-29:
First, isolate your waypoints, shift them, and save:
gpsbabel -i ... -f ... -x nuketypes=tracks,routes -x transform,trk=wpt, -x track,move=7168d -x transform ,wpt=trk -o ... -F my_shifted_waypoints....
Next, isolate your tracks, shift them, and save
gpsbabel -i ... -f ... -x nuketypes=waypoints,routes -x track,move=7168d, -o ... -F my_shifted_tracks...
Next, combine your shifted tracks and waypoints.
gpsbabel -i ... -f my_shifted_waypoints... -f my_shifted_tracks... -o ... -F combined_shifted_output...

MAlfare, 2019-12-29:
Than you for your tips! They came very quickly but at 3 a.m. I did not want to try them immediately ;-)
As mtkbabel can produce, beside bin-, also gpx-files with tracks, waypoints and both, I could save the nuking. An other approach would be the use of sed, what would need some time to get into.
Thanks again for your quick support!

MAlfare, 2019-12-29:
Hi tsteven4 again.
I could do it this way:
gpsbabel -i gpx -f "$O_FILENAME".gpx -x nuketypes,tracks,routes -x transform,trk=wpt -x nuketypes,waypoints -x track,move=7169d -x transform,wpt=trk \
-x nuketypes,tracks -o gpx,gpxver=1.1 -F "$O_FILENAME"-WP.gpx

(2 additonal nukes to inhibit doubling waypoints (ones with old date and ones with new date), and a track of waypoints.)

gpsbabel -i gpx -f "$O_FILENAME".gpx -x nuketypes,waypoints,routes -x track,move=7169d, -o gpx,gpxver=1.1 -F "$O_FILENAME"-TR.gpx

gpsbabel -i gpx -f "$O_FILENAME"-TR.gpx -f "$O_FILENAME"-WP.gpx -o gpx,gpxver=1.1 -F "$O_FILENAME"-OK.gpx

I had to replace "nuketypes=" by"nuketypes," ;-)
And yes, gpsbabel CAN read tracks AND waypoints from the m241 Device, at least in CLI, I beg gpsbabels pardon ;-)

tsteven4, 2019-12-29:
> (2 additonal nukes to inhibit doubling waypoints (ones with old date and ones with new date), and a track of waypoints.)

or use transform del option 
https://www.gpsbabel.org/htmldoc-development/filter_transform.html

MAlfare, 2019-12-29:
Hi!
I have learned a lot today/-night, but the lot I have to, is larger. ;-)
Thank you again.