GPSBabel/gpsbabel #474: xcsv style derived formats may have wrong GUI and/or CLI capability

tsteven4, 2020-01-25:
The following formats style files don't have the optional DATATYPE directive.  Due to a bug the capability passed to the GUI is either zero (unknown which translates to waypoint) if no styles that proceed the format in the style list have used a DATATYPE directive, or the DATATYPE of the most recent preceding format in the style list.

The document states 
> By default all data are seen as waypoint data.

but you can see this is rarely the behavior with the GUI.

Sytle based format without DATATYPE | Accidental GUI capability
-- | --
arc | track
cambridge | track
csv | track
cup | track
custom | track
dna | track
flysight | track
fugawi | track
garmin301 | track
garmin_poi | track
geonet | track
gpsdrive | track
gpsdrivetrack | track
gpsman | track
mapconverter | track
motoactv | track
mxf | track
navigonwpt | track
nima | track
openoffice | track
s_and_t | waypoint
saplus | waypoint
tabsep | waypoint
xmap2006 | Unkonwn => waypoint
xmap | Unkonwn => waypoint
xmapwpt | Unkonwn => waypoint

With the CLI, the capability depends on the capability of any style based format on the command line that proceeds the format in question.  If no style based formats proceed the format in question then the capability is unknown, which gets treated as waypoint.
For example with this command reads all the points as tracks
`./gpsbabel -i iblue747 -f reference/track/iblue747.csv -i csv -f x.csv -o gpx -F -`
but this command reads the points as waypoints.
`./gpsbabel  -i csv -f x.csv -o gpx -F -`

robertlipe, 2020-01-25:
Gack!

The following formats style files don't have the optional DATATYPE
> directive. Due to a bug the capability passed to the GUI is either
>
DATATYPE was an afterthought in style handling. (I didn't think it would be
useful because surely a file could contain multiple tracks or routes and
without headers to mark where they started and ended, that was nonsensical.
That was short-sighted because there are times - like arc - where sometimes
a list of coords makese sense to interpret as a track.

> zero (unknown which translates to waypoint) if no styles that proceed the
> format in the style list have used a DATATYPE directive, or the DATATYPE of
> the most recent preceding format in the style list.
>
"Clearly" that's a defect. Retaining global state between formats is bad.
That's one of the reasons I've long fantasized about moving most format
globals into member variables and having a Format created at -f or -F time.
(Maybe -i or -o...)

> The document states
>
> By default all data are seen as waypoint data.
>
> We might think about making this mandatory in the style file.

Many of these are old and I imagine unused. Mapoplis.com, for example, has
fallen to domain squatters.

I can run through this list and declare formats involving motion
(gpsdriveTRACK or FLYsight) to be tracks.  Some like arc and csv seem
closer to the border, but let's declare them to officially be what the doc
said we'd be, waypoints.

I'll handle this tonight if you don't already have a patch in the wings.

> With the CLI, the capability depends on the capability of any style based
> format on the command line that proceeds the format in question. If no
> style based formats proceed the format in question then the capability is
> unknown, which gets treated as waypoint.
>
Darned conditionally-set globals. :-(

RJL

tsteven4, 2020-01-25:
A fixed branch has already been pushed to a fork, a PR should appear soon.