GPSBabel/gpsbabel #482: Order of LineString and TimeSpan in kml files

Stadly, 2020-01-28:
In kml files, when the `LineString` element precedes the `TimeSpan` element, the track points don't get timestamps when converting to gpx format, as implemented in https://github.com/gpsbabel/gpsbabel/pull/115.

I am not sure if `LineString` preceding `TimeSpan` makes for an invalid kml file, but it is what you get if you do "Export this day to KML" in [Google Maps Timeline](https://www.google.com/maps/timeline), so it would be really nice if it would work :)

For instance this kml file:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<kml xmlns="http://www.opengis.net/kml/2.2" 
   xmlns:gx="http://www.google.com/kml/ext/2.2">
   <Document>
      <Placemark>
         <LineString>
            <coordinates>10.7602892,59.929845900000004 10.7602892,59.929845900000004 10.7592405,59.9288833</coordinates>
         </LineString>
         <TimeSpan>
            <begin>2019-05-01T10:00:22.000Z</begin>
            <end>2019-05-01T10:42:13.726Z</end>
         </TimeSpan>
      </Placemark>
   </Document>
</kml>
```
Does not have timestamps when converted to gpx:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<gpx version="1.0" creator="GPSBabel - http://www.gpsbabel.org" xmlns="http://www.topografix.com/GPX/1/0">
  <time>2020-01-28T20:16:44.096Z</time>
  <bounds minlat="59.928883300" minlon="10.759240500" maxlat="59.929845900" maxlon="10.760289200"/>
  <trk>
    <trkseg>
      <trkpt lat="59.929845900" lon="10.760289200"/>
      <trkpt lat="59.929845900" lon="10.760289200"/>
      <trkpt lat="59.928883300" lon="10.759240500"/>
    </trkseg>
  </trk>
</gpx>
```

While this kml file:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<kml xmlns="http://www.opengis.net/kml/2.2" 
   xmlns:gx="http://www.google.com/kml/ext/2.2">
   <Document>
      <Placemark>
         <TimeSpan>
            <begin>2019-05-01T10:00:22.000Z</begin>
            <end>2019-05-01T10:42:13.726Z</end>
         </TimeSpan>
         <LineString>
            <coordinates>10.7602892,59.929845900000004 10.7602892,59.929845900000004 10.7592405,59.9288833</coordinates>
         </LineString>
      </Placemark>
   </Document>
</kml>
```
Does have timestamps when converted to gpx:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<gpx version="1.0" creator="GPSBabel - http://www.gpsbabel.org" xmlns="http://www.topografix.com/GPX/1/0">
  <time>2020-01-28T20:16:13.198Z</time>
  <bounds minlat="59.928883300" minlon="10.759240500" maxlat="59.929845900" maxlon="10.760289200"/>
  <trk>
    <trkseg>
      <trkpt lat="59.929845900" lon="10.760289200">
        <time>2019-05-01T10:00:22Z</time>
      </trkpt>
      <trkpt lat="59.929845900" lon="10.760289200">
        <time>2019-05-01T10:14:19.242Z</time>
      </trkpt>
      <trkpt lat="59.928883300" lon="10.759240500">
        <time>2019-05-01T10:28:16.484Z</time>
      </trkpt>
    </trkseg>
  </trk>
</gpx>
```

tsteven4, 2020-01-28:
Your first sample, quoted below, violates the  kml 2.2 schema. If Google 
Maps Timeline created this then they have a bug.

On 1/28/2020 1:21 PM, Stadly wrote:
> <?xmlversion="1.0"encoding="UTF-8"?>
> <kml  xmlns="http://www.opengis.net/kml/2.2"
>     xmlns:gx="http://www.google.com/kml/ext/2.2">
>     <Document>
>        <Placemark> [...]

You second sample is valid kml:

> <?xmlversion="1.0"encoding="UTF-8"?>
> <kml  xmlns="http://www.opengis.net/kml/2.2"
>     xmlns:gx="http://www.google.com/kml/ext/2.2">
>     <Document>
>        <Placemark> [...]

Stadly, 2020-01-28:
Ok, thanks. Then there is a bug in Google Maps Timeline :-/ Not sure how easy it will be to get them to fix it. Would it be possible to let GPSBabel understand such malformed data?

Stadly, 2020-01-28:
I have sent feedback to Google concerning the bug.

Stadly, 2020-01-29:
I noticed that both the following kml files result in the same gpx file, containing the timestamp, even though the first one is invalid (`Point` before `TimeStamp`) while the second one is valid (`TimeStamp` before `Point`).

```xml
<?xml version="1.0" encoding="UTF-8"?>
<kml xmlns="http://www.opengis.net/kml/2.2" 
   xmlns:gx="http://www.google.com/kml/ext/2.2">
   <Document>
      <Placemark>
         <Point>
            <coordinates>10.760289199999999,59.9298459</coordinates>
         </Point>
         <TimeStamp>
            <when>2019-04-30T20:35:42.388Z</when>
         </TimeStamp>
      </Placemark>
   </Document>
</kml>
```

```xml
<?xml version="1.0" encoding="UTF-8"?>
<kml xmlns="http://www.opengis.net/kml/2.2" 
   xmlns:gx="http://www.google.com/kml/ext/2.2">
   <Document>
      <Placemark>
         <TimeStamp>
            <when>2019-04-30T20:35:42.388Z</when>
         </TimeStamp>
         <Point>
            <coordinates>10.760289199999999,59.9298459</coordinates>
         </Point>
      </Placemark>
   </Document>
</kml>
```

So GPSBabel successfully parses `TimeStamp` following `Point`. It would be nice if it similarly would successfully parse `TimeSpan` following `LineString`.

gpsbabel, 2020-05-20:
This a well documented report. Thank you. I've looked at the code in question and while I agree it would be nice if we could always divine a sensible reading of malformed input, it's just not practical in all cases.

Also, shame on Google. In addition to creating malformed input, this should probably be gx:Track instead of using TimeSpan anyway. It has a much better way of interpolating needed data.

closed 2020-05-20 by gpsbabel

maximecharriere, 2022-05-24:
Hello @Stadly,
I'm trying to do exactly the same thing as you :  
1. `Export this day to KML` in Google Maps Timeline
2. Convert kml to GPX with GPSBabel
3. Import GPX in Adobe Lightroom to add location to my photos

However Lightroom need the timespan to link the photo with the location.  And due to the Google export error, GPSBabel can't get the timespan.  

That was two years ago, but did you remember how you did it ?  
Did you get a reply from Google ?

Related issue : #484 #516

Stadly, 2022-05-26:
@maximecharriere I ended up writing a script that downloads KML files from Google Maps Timeline and then fixes the issues before saving. It is all part of this project: https://github.com/Stadly/TimelineExtractor Hope it helps :)