GPSBabel/gpsbabel #493: arc-project.test fails due to different time result

cjmayo, 2020-02-04:
This is gpsbabel 1.6.0 on Gentoo Linux:
```diff
Running ./testo.d/arc-project.test
--- ./reference/arc-project3.gpx
+++ /tmp/gpsbabel.29/arc-project3.gpx
@@ -3,7 +3,7 @@
   <time>1970-01-01T00:00:00Z</time>
   <bounds minlat="28.353803737" minlon="-16.502955209" maxlat="28.359546000" maxlon="-16.494318496"/>
   <wpt lat="28.357846000" lon="-16.499208000">
-    <time>2011-09-26T00:00:51Z</time>
+    <time>2011-09-26T00:00:50.999Z</time>
     <name>Derecha</name>
     <cmt>Derecha</cmt>
     <desc>Derecha</desc>
ERROR comparing ./reference/arc-project3.gpx /tmp/gpsbabel.29/arc-project3.gpx
--- ./reference/arc-project3.gpx
+++ /tmp/gpsbabel.29/arc-project3.gpx
@@ -3,7 +3,7 @@
   <time>1970-01-01T00:00:00Z</time>
   <bounds minlat="28.353803737" minlon="-16.502955209" maxlat="28.359546000" maxlon="-16.494318496"/>
   <wpt lat="28.357846000" lon="-16.499208000">
-    <time>2011-09-26T00:00:51Z</time>
+    <time>2011-09-26T00:00:50.999Z</time>
     <name>Derecha</name>
     <cmt>Derecha</cmt>
     <desc>Derecha</desc>
ERROR comparing ./reference/arc-project3.gpx /tmp/gpsbabel.29/arc-project3.gpx
```

tsteven4, 2020-02-04:
@cjmayo I think I may see the issue.  If you can verify the HEAD still fails I'll try a change.

cjmayo, 2020-02-05:
Same test failure with HEAD (9f5e3e4b97fb29a3d63673ed7e2d9ba946989833).

After patching with #495 the test passes - thanks!

tsteven4, 2020-02-05:
Thanks for the report and testing.

PR #495 (2020-02-05): use rounding when interpolating time in arcdist filter.
It is hoped that this fixes #493.

eliminate unnecessary casting away of const.
reduce variable scope.
closed 2020-02-05 by tsteven4