GPSBabel/gpsbabel #516: Incorrect conversion from kml to gpx

Stadly, 2020-03-05:
Converting this `kml` file to `gpx`:
```xml
<kml xmlns="http://www.opengis.net/kml/2.2">
    <Document>
        <Placemark>
            <TimeSpan>
                <begin>2019-05-01T12:00:00Z</begin>
                <end>2019-05-01T14:00:00Z</end>
            </TimeSpan>
            <LineString>
                <coordinates>1,2,3 4,5,6 7,8,9</coordinates>
            </LineString>
        </Placemark>
    </Document>
</kml>
```

Yields this result:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<gpx version="1.0" creator="GPSBabel - http://www.gpsbabel.org" xmlns="http://www.topografix.com/GPX/1/0">
  <time>2020-03-05T22:20:49.777Z</time>
  <bounds minlat="2.000000000" minlon="1.000000000" maxlat="8.000000000" maxlon="7.000000000"/>
  <trk>
    <trkseg>
      <trkpt lat="2.000000000" lon="1.000000000">
        <ele>3.000</ele>
        <time>2019-05-01T12:00:00Z</time>
      </trkpt>
      <trkpt lat="5.000000000" lon="4.000000000">
        <ele>6.000</ele>
        <time>2019-05-01T12:40:00Z</time>
      </trkpt>
      <trkpt lat="8.000000000" lon="7.000000000">
        <ele>9.000</ele>
        <time>2019-05-01T13:20:00Z</time>
      </trkpt>
    </trkseg>
  </trk>
</gpx>
```

As you can see, the `time`s for the `trkpt`s are incorrect. I would have expected the times for the `trkpt`s to be 12:00, 13:00 and 14:00:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<gpx version="1.0" creator="GPSBabel - http://www.gpsbabel.org" xmlns="http://www.topografix.com/GPX/1/0">
  <time>2020-03-05T22:20:49.777Z</time>
  <bounds minlat="2.000000000" minlon="1.000000000" maxlat="8.000000000" maxlon="7.000000000"/>
  <trk>
    <trkseg>
      <trkpt lat="2.000000000" lon="1.000000000">
        <ele>3.000</ele>
        <time>2019-05-01T12:00:00Z</time>
      </trkpt>
      <trkpt lat="5.000000000" lon="4.000000000">
        <ele>6.000</ele>
        <time>2019-05-01T13:00:00Z</time>
      </trkpt>
      <trkpt lat="8.000000000" lon="7.000000000">
        <ele>9.000</ele>
        <time>2019-05-01T14:00:00Z</time>
      </trkpt>
    </trkseg>
  </trk>
</gpx>
```

Currently, if there are `x` coordinates, the time span is divided into `x+1` points in time instead of just `x`, and there are no coordinates for the last one.

robertlipe, 2020-03-13:
Thank you for a well-described bugreport with a useful test case.

It's easy enough to get the result that you want just edit kml.cc :
      qint64 ms_per_waypoint = timespan_ms / trk_head->rte_waypt_ct;
to
      qint64 ms_per_waypoint = timespan_ms / (trk_head->rte_waypt_ct -1);

However, I'm less certain that it's actually *correct* by the KML spec.
Does that TimeSpan actually include both start and end or does it mean
"right up to the end"? In that case, the "right" answer would have an
obnoxious amount of decimal noise.

I think the colloquial "fencepost" definition here might be the best one.
If you count five fence posts, you have four fence rails, right?
1 -post1- 2 -post2- 3 -post3- 4 -post4- 5

If you can confirm that this fix works for you, I'll commit it and quit
obsessing about finding OGC blessing for that reading of the spec.
RJL

Stadly, 2020-03-13:
I have not built GPSBabel to test it myself, but from reading the source code, that change should definitely do the trick. Thanks!

I have not had a look at the kml spec, but from common sense I think the change you suggest would be correct. If you can point me to a particular section of the spec I can try to decipher it to see if we've missed something.

gpsbabel, 2020-03-13:
Thanx. Let's just go with it. I also added an error check to avoid a divide
by zero if you try to reduce a "path" with only one point.

Fix is queued up in
https://ci.appveyor.com/project/gpsbabel/gpsbabel/builds/31441405

...after I tidy up some other commits that have been in the oven for (too)
many weeks.

robertlipe, 2020-03-13:
The fix is https://github.com/gpsbabel/gpsbabel/pull/518 Please review.

Stadly, 2020-03-14:
Looks good to me. Thanks!

closed 2020-03-14 by Stadly