GPSBabel/gpsbabel #517: dg-100 regression: download fails with checksum mismatch

akhuettel, 2020-03-12:
I recently re-activated my old and trusty DG-100 logger, but now I can't download recorded data anymore. My issue (on Gentoo linux) seems to be identical with Debian bug 804476, 
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=804476

If that is the case, then the bug was introduced somewhere between 1.5.0 and 1.5.2.

huettel@noumea ~ $ gpsbabel -t -w -D 1 -i dg-100 -f /dev/ttyUSB0 -o gpx -F /home/huettel/ki-test.gpx
GPSBabel Version: 1.5.4 
options: module/option=value: dg-100/erase="0" (=default)
options: module/option=value: dg-100/erase_only="0" (=default)
TX: Frame Start a0 a2 Payload_Len 0003 Cmd: getfileheader
RX: Start a0a2 Len 0016 Cmd: getfileheader
found 1 headers, nextheader=1
Header #0: Seq: 0 Time: Thu Mar 12 14:47:30 2020
TX: Frame Start a0 a2 Payload_Len 0003 Cmd: getfileheader
RX: Start a0a2 Len 000a Cmd: getfileheader
found 0 headers, nextheader=0
no further headers, aborting the loop
TX: Frame Start a0 a2 Payload_Len 0003 Cmd: getfile
RX: Start a0a2 Len 0401 Cmd: getfile
RX: Start a0a2 Len 0401 Cmd: getfile
TX: Frame Start a0 a2 Payload_Len 0001 Cmd: getconfig
RX: Start a0a2 Len 0039 Cmd: getconfig
checksum mismatch: data sum is 0x043c, checksum received is 0x0004
huettel@noumea ~ $

akhuettel, 2020-03-12:
Reverting dg-100.cc to the 1.5.0 state (and re-applying some type fixes so things build again) makes the command work:

huettel@noumea ~ $ gpsbabel -t -w -D 1 -i dg-100 -f /dev/ttyUSB0 -o gpx -F /home/huettel/ki-test.gpx
GPSBabel Version: 1.5.4 
options: module/option=value: dg-100/erase="0" (=default)
options: module/option=value: dg-100/erase_only="0" (=default)
TX: Frame Start a0 a2 Payload_Len 0003 Cmd: getfileheader
RX: Start a0a2 Len 0016 Cmd: getfileheader
found 1 headers, nextheader=1
Header #0: Seq: 0 Time: Thu Mar 12 14:47:30 2020
TX: Frame Start a0 a2 Payload_Len 0003 Cmd: getfileheader
RX: Start a0a2 Len 000a Cmd: getfileheader
found 0 headers, nextheader=0
no further headers, aborting the loop
TX: Frame Start a0 a2 Payload_Len 0003 Cmd: getfile
RX: Start a0a2 Len 0401 Cmd: getfile
RX: Start a0a2 Len 0401 Cmd: getfile
cet_util: Converting from "US-ASCII" to "UTF-8", done.
options: module/option=value: gpx/snlen="32" (=default)
huettel@noumea ~ $

tsteven4, 2020-03-12:
Can you try this with -D 5 and report the output?  I am wondering if the 
checksum is being computed over the wrong length of data.

Can you build with https://github.com/gpsbabel/gpsbabel and try that?

akhuettel, 2020-03-12:
> Can you try this with -D 5 and report the output?  I am wondering if the
checksum is being computed over the wrong length of data.

Log is attached (1.5.4 displaying the error)
[gpsbabel-1.5.4-error.log](https://github.com/gpsbabel/gpsbabel/files/4326562/gpsbabel-1.5.4-error.log)

akhuettel, 2020-03-12:
> Can you build with https://github.com/gpsbabel/gpsbabel and try that?

Today's git master shows the same problem. Log is attached.
[gpsbabel-master-error.log](https://github.com/gpsbabel/gpsbabel/files/4326587/gpsbabel-master-error.log)

tsteven4, 2020-03-14:
The submitted log shows that the getconfig command is responding with one more byte than we expect.  This results in our grabbing the checksum from the wrong two bytes, which results in the checksum mismatch and fatal error.  The checksum does appear to be correct if we get it from the correct place.
However, there is evidence in the comments and on the web of the DG-100 returning a different number of bytes than your sample.  Furthermore, there are widespread indications that the payload length field should be ignored on return data.  This makes it quite difficult to parse any return data that isn't fixed in size.  To make matters worse the references for the protocol have largely disappeared from the web.
It seems that the getconfig command was added in release 1.5.1 to change the status light to green on the DG-200.  We never have done anything with any data returned from the getconfig command.
I will submit a PR that will not issue the getconfig command on DG-100 devices.  This will restore the getconfig behavior for the DG-100 to 1.5.0.

tsteven4, 2020-03-14:
Please try #519

tsteven4, 2020-03-17:
@akhuettel , can you please try #519?

akhuettel, 2020-03-26:
I can confirm that the patch works fine on top of version 1.5.4 - I can download the data again. Thanks! (and since that part of the code hasnt changed, it should work with master too)

tsteven4, 2020-03-26:
Fixed in #519

tsteven4, 2020-03-26:
@akhuettel can you apply the patch and close https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=804476?

closed 2020-03-26 by tsteven4

amic-github, 2020-05-02:
It worked indeed, but I have trouble using erase and erase_only options. Reverting back to 1.5.0 solved the issue. I may try to investigate on that to see in which version the problem was introduced. @akhuettel could you tell me if you have the same problem ?

tsteven4, 2020-05-05:
@akhuettel 
I have translated the received bytes from your log gpsbabel-1.5.4-error.log to a binary file and modified dg-100.cc to allow reading of this file.  This allows us to test this module in regression.  The test doesn't use the serial device so coverage isn't complete, but something is a lot better than what we have now (nothing).  Are you OK with creating a reference file from your log?  The track would become part of our repository in a binary format as well as a gpx file.

@akhuettel , @amic-github 
- can either of you test a dg-100 or dg-200 device after I make the above changes to make sure the device io still works?
- can either of you submit a log from a dg-200 device like gpsbabel-1.5.4-error.log? Then I could add another test for the dg-200 device.