GPSBabel/gpsbabel #527: No way to eliminate KML <Style>

jidanni, 2020-03-28:
For -o kml, there is no way to eliminate the <Style>, and just use the
(viewer software, e.g., Google Earth) defaults. At most one can only use
deficon="file://myicon.png" to change it.

```
      <Placemark>
        <name>411944-24</name>
        <TimeStamp>
          <when>2020-01-17T14:40:02.550Z</when>
        </TimeStamp>
        <Style>
          <IconStyle>
            <Icon>
              <href>file://myicon.png</href>
            </Icon>
          </IconStyle>
        </Style>
        <Point>
          <coordinates>120.701952,24.159254,134.00</coordinates>
        </Point>
      </Placemark>

```
One must resort to crude after-filters:
`sed '/<Style>/,/<\/Style>/d; /<styleUrl>/d;'`

robertlipe, 2020-03-29:
Earth is but one reader. This ensures a consistent style in Maps, ArcGIS,
Grass, and so on.

We're really not an editor/styling tool. We're not going to add more
command line flags to allow you to create your own. If you have custom
needs that go beyond conversion, you should look at other tools or maybe
even hack up our KML writer to more suit you.

RJL

jidanni, 2020-03-29:
> Earth is but one reader. This ensures a consistent style in Maps, ArcGIS, Grass, and so on. 

Sure, if one is online. But the hardwiring of http(s) resources into the output causes all sorts of disasters offline. E.g., big X's instead of the KML browser's default in-built icons.

> We're really not an editor/styling tool.

Yes you are. There was no style in the source input. You put the Style in.

The KML was guaranteed to work in Maps, ArcGIS, Grass, and so on. 
Now with the style you put in, it will only work with a network connection.
Even worse is you are not just saying "Use a yellow marker", you are hardwiring https paths that
have been known to break in the past.

One might think that deficon="" might finally stop it, but it doesn't!

gpsbabel, 2020-05-19:
I'd accept a path that disabled writing the style in KML but that cannot be our default. Sounds like you'll probably be happier with your sed tweak or a locally modified copy of GPSBabel.

We'll disagree on whether we're a style editor.

closed 2020-05-19 by gpsbabel