GPSBabel/gpsbabel #533: xcsv style: date overwrites time (both IFIELD GMT_TIME)

kaefert, 2020-04-09:
Hi there!

I am trying to convert a csv file that has seperate columns for time and date, and it seems to me the later appearance of the date which is are both defined using the IFIELD GMT_TIME:
[kdrcsv.style.txt](https://github.com/gpsbabel/gpsbabel/files/4456495/kdrcsv.style.txt)

Does this need to be defined differently in the xcsv style or is this a bug in gpsbabel?

Thank you for the great piece of software!

tsteven4, 2020-04-09:
You didn't mention what version of gpsbabel you are using.

I acknowledge it doesn't do what you want.  The specification https://www.gpsbabel.org/htmldoc-development/style_define.html for GMT_TIME and LOCAL_TIME makes no mention that these fields can be used multiple times and the values accumulated.  However, LOCAL_TIME does this and GMT_TIME doesn't.  The spec does mention HMSG_TIME and HMSL_TIME get added to a previously  parsed *_TIME field, but 1) your time precedes your date.  Notice that the format for HMS*_TIME doesn't use strptime conversion format tags like GMT_TIME and LOCAL_TIME.

Something to consider that makes it behave like you want is in the patch below.  It makes GMT_TIME and LOCAL_TIME behave consistently, they both accumulate.  The specification should mention this already for LOCAL_TIME, and if we adopt this behavior for GMT_TIME as well.

[patch.txt](https://github.com/gpsbabel/gpsbabel/files/4457050/patch.txt)

kaefert, 2020-04-11:
Hey there! Sorry for not telling about the gpsbabel version I've been using.

I'm on a Manjaro KDE (arch based) system, having installed GPSBabel 1.6.0-2 from the repository.
To try out your patch I guess I'll have to see how to compile it myself.

There's an [AUR package](http://aur.archlinux.org/packages/gebabbel) for that, though it's from 2019-01-05 and uses sources from [Sourceforge](http://gebabbel.sourceforge.net/) so I guess this won't do (as is)..

Thanks for the hint with LOCAL_TIME, I tried that and it works fine, it's only shifted by some hours, but better than no time at all :)

kaefert, 2020-04-12:
okey so the build process was really simple, great work!

```
git clone https://github.com/gpsbabel/gpsbabel.git
cd gpsbabel
./configure
make 
```

Now I just need to find out how to apply a patch and repeat the `make` step.

tsteven4, 2020-04-12:
git apply patch.txt
make

If your working from the command line you are done, i.e. just run gpsbabel.

If you want to build the GUI
make linux-gui

Then, in gui/GPSBabelFE run gpsbabelfe.
The GUI won't be portable because it isn't packaged with the Qt libraries, but it should work on your build system.

kaefert, 2020-04-12:
okey managed to test the build now, worked fine with the patch and the style reverted to GMT_TIME fields :)
Thank you for the help! Would love if you could merge it into the repository so that future releases will work like that out of the box :)

gpsbabel, 2020-05-19:
User confirmed fix is working. Thanx, all. Closing.

closed 2020-05-19 by gpsbabel