GPSBabel/gpsbabel #534: subRip subtitle datetime options aren't optional

Matthias84, 2020-04-10:
Hi, I encounter a lot of troubles, while getting a working setup to georeference a video sequence with a GPS track.
Long story short: The gps_date setting isn't optional, [like the docs say](https://www.gpsbabel.org/htmldoc-development/fmt_subrip.html).
Currently if you run it only with gps_time (because you expect it be optional, as it is just one day), it will fall back to start always at the very first GPS node, ignoring your gps_time
`gpsbabel -i gpx -f tracklog.gpx -o subrip,video_time=000142,gps_time=192521 -F subtitles.srt`

IMHO this is unexpected and pretty confusing. I recommend to print at least a warning, that we fallback to this default behaviour. Otherwise indicating both parameters as non optional, will avoid this misunderstanding?
Another idea is to use just the very first day of a GPS track as default, so in most cases it doesn't need the extra gps_date setting?

P.S: At the GUI (subrip options dialog) the gps_date label is wrong. The format string is 'yyyymmdd'

tsteven4, 2020-04-10:
Thanks for your report, also the postscript.  I am sorry for your difficulties.

I note there is a comment in the code https://github.com/gpsbabel/gpsbabel/blob/a66e6468bd7f04eb4c3984ce4b404259d5329d9b/subrip.cc#L264

It seems this was done:
In the document https://www.gpsbabel.org/htmldoc-development/fmt_subrip.html under gps_time is states 

> This option must be used together with gps_date; if one or both are missing, the timestamp of the first GPS trackpoint is used. 

Under gps_date it states 

> This option must be used together with gps_time; if one or both are missing, the timestamp of the first GPS trackpoint is used.

What part of the document led you to believe you could use one without the other?

Printing a warning is reasonable.  I don't understand what you mean by "Otherwise indicating both parameters as non optional".   Using the first day of the track as a default could be confusing as the date could easily roll over, especially since the times are in UTC.

Matthias84, 2020-04-10:
Hi, thanks for the quick response :+1: 

if you just fire up the GUI, you are allowed to use one field or the other :wink: So I just expected, that you get an appropriate date otherwise. As the processing finishes without troubles (and in some constellations with a good result), it was hard to see what is the problem.

A warning could be a hint, just mentioning the default fallback, or a error message, showing that you will need both params to get a useful. Last one was the "inidicating non optional parameters".

I agree, that guessing the date isn't trivial. On the other hand, you usually use passpoints at start or end of a GPS track, so maybe it's not that bad?

Anyway, thank you all for keeping working on gpsbabel and have nice holidays :tada:

tsteven4, 2020-04-10:
@Matthias84 can you build and test code if I make it available on github somewhere?

Alternatively, I can point you to a windows installer to try if your using windows.

Matthias84, 2020-04-11:
@tsteven4 A  win64 executable would be great :+1:

tsteven4, 2020-04-11:
@Matthias84 please try https://ci.appveyor.com/project/tsteven4/gpsbabel/builds/32112213/job/ysdq2fgq9kp83edg/artifacts.  If your testing is successful I will submit this as a pull request.

If you add the debug option at level 2 or higher (-D2 on the command line) you will see messages about synchronization.

Subsecond synchronization is supported now, i.e. you can use hhmmss or hhmmss.sss.

GUI description should now be correct (yyyymmdd).

Invalid option input for gps_time, gps_date and video_time should cause a fatal error with an explanation.

An attempt to use gps_time without gps_date should cause a fatal error.
An attempt to use gps_date without gps_time should cause a fatal error.
Unfortunately the handoff between the command line executable and the GUI executable is not sophisticated enough to communicate to the GUI that both or neither option is required, so it is still possible to enter only one on the GUI.  But this will result in a fatal error with an explanation, so you can correct it instead of wondering why your video isn't aligned.

Please check this and verify that the synchronization is as you expect.

Thanks

Matthias84, 2020-04-13:
Hi, I tested it and it works fine! Thank you very much :smile:

tsteven4, 2020-04-13:
This is fixed in #536

gpsbabel, 2020-05-19:
User has confirmed the fix worked. Thank you, tsteven4.

closed 2020-05-19 by gpsbabel