GPSBabel/gpsbabel #535: No /dev/ttyACM0 conection

jvmax, 2020-04-12:
I'm trying to retrieve data from a Qstarz BT-Q1000XT which connects to /dev/ttyACM0.  That's not an option on gpsbabel gui and I get no response when I try from the command line.  Is it possible to connect to /dev/ttyACM0?

tsteven4, 2020-04-13:
Be aware that #515 is adding support for this device.  However, it is file based instead of device based.

>  Handle Qstarz BL-1000 .BIN files.
    Saved in the SESSION/GPSLog folder. There will 3 files sharing the same base name: .BIN, .POI, and .DAT.
    Only the .BIN file is of interest to us.

tsteven4, 2020-04-18:
@jvmax Does the newly merged qstarz_bl-1000 format work for you?  For Linux you will need to build from master branch.

> Qstarz BL-1000 (qstarz_bl-1000)
This format can...
• read waypoints
• read tracks
This format is used by the Qstarz BL-1000GT and BL-1000ST.
These Qstarz devices are file based. GPS logs are saved as *.BIN files to the SESSION/GPSLog folder.

gpsbabel, 2020-05-19:
User has gone non-responsive. Closing.

closed 2020-05-19 by gpsbabel