GPSBabel/gpsbabel #539: Garmin Tactix Delta

nemanjan00, 2020-04-15:
Hi guys,

I found nothing googleing this error, so, I shared it here, just in case someone has any ideas. 

Should Garmin Tactix Delta even be supported? 

``` bash
$ gpsbabel -D 9 -i garmin -f /dev/ttyUSB0 -o gpx -F foo.gpx
GPSBabel Version: 1.6.0 
main: Compiled with Qt 5.13.0 for architecture x86_64-little_endian-lp64
main: Running with Qt 5.14.2 on Arch Linux, x86_64
main: QLocale::system() is en_US
main: QLocale() is en_US
main: QTextCodec::codecForLocale() is UTF-8, mib 106
GPS Serial Open at 9600
Tx Data:10 fe 00 02 10 03 : ...(PRDREQ  )
Rx Data:10 06 02 fe 00 fa 10 03  .. (ACK     )
Rx Data:10 f8 0a 47 50 53 20 56 34 2e 31 30 00 db 10 03  GPS.V4.10. (PRDEDA  )
Tx Data:10 06 02 f8 00 00 10 03 : .....(ACK     )
Unit:	V4.10
ID:	20551
Version:	82.75[WARNING] A001 protocol not supported
[ERROR] INIT: No table entry for ID 20551
```

robertlipe, 2020-04-15:
Hi and welcome.

I don't have one of these and I'm not a Garmin PM, so I'll venture a guess
of "maybe, but not like that".

The Garmin USB/Serial protocol that you're trying to use hasn't really been
used by any NEW Garmin designs in almost fifteen years. Everything about
location above is empty. For example, everything depends on A001.

Most of of the fitness products for the last 10-12 years show up on USB as
a mass storage (disk) device that let you copy TCX, FIT,, or GPX tracks
from there. We're able to read almost all those, but we're about puttin
your run (or ride or whatever) on a map and not a fitness app. We
primarily convert location to other formats and try to bring along some
simple data elements when we can. We're not, say, Stratva, trying to
compute calories burned based on elevation and temperature. If that data is
in the formats in a way we know how to preserve, we'll try to drag it out,
but we won't compute it.

Look for a way to make your device show up like. a thumb drive. From there,
we bay be able to pull the FIT or TCS files out and do something usable
with them.

Please report back with your finding. I'll put a footnote (that everyone
will miss in that ocean of Garmin models on that page...sigh.) in the
fmt_garmin page. If you can include a brief paragraph on how to get to this
mode and make it work, even better.

Thanx,
RJL

nemanjan00, 2020-04-19:
Thanks a lot,

Also, I am attempting to make my watch work as external GPS, for gpsd. 

Any info on that?

robertlipe, 2020-04-20:
They've been dropping that feature for years. Even their products meant for
automotive use don't support NEMA or PVT to announce location + time, such
as is the common use for GPSd. That stopped about the mid 200x decade. If
your device supported it, there would be a PVT entry in the -D9 debug entry
or a NMEA serial option in the settings.

Fortunately, bluetooth or USB/Serial "puck" products that burp up location
are dirt cheap.

Good luck
RJL

nemanjan00, 2020-04-21:
Ok, 

You have answered all of the questions I had. :D

Hopefuly this will be helpfull to other people too... 

Nice to know there are people who are devotet so much to opensource project, to know so much about niche products like GPS devices :D

gpsbabel, 2020-05-19:
You're welcome. Good luck.

closed 2020-05-19 by gpsbabel