GPSBabel/gpsbabel #571: configure: error: libusb-1.0 not found (--without-libusb to disable)

rickrich, 2020-05-25:
$ ./configure 
...
checking if Qt version >= 5.9... yes
checking for libusb_init in -lusb-1.0... no
configure: error: in `/home/rick/proj/gpsbabel':
configure: error: libusb-1.0 not found (--without-libusb to disable)
See `config.log' for more details

tsteven4, 2020-05-25:
That is exactly what was intended to happen when, as the error says, libusb-1.0 isn't found.  We had hoped the error message would make it clear you either need to install libusb-1.0 (and actually the dev package so we get the h file), or accept reduced functionality and disable libusb-1.0.  The error message explicitly directs you how to do that.

libusb-1.0 is a new dependency relative to gpsbabel release 1.6.0.  We used to have a dependency on libusb-0.1 that no longer exists.  We intentionally chose to not silently build without libusb-1.0 as we were afraid packagers would unintentionally create packages with reduced functionality.  But we hoped the error message was self explanatory, it is almost verbatim out of the autoconf AC_ARG_WITH man page.

For ubuntu the package is libusb-1.0.0-dev.

rickrich, 2020-05-25:
What is the Fedora package?

tsteven4, 2020-05-25:
Looks like libusbx-devel.
https://apps.fedoraproject.org/packages/libusbx-devel

We don't do a CI build on fedora, so let us know how it goes.

rickrich, 2020-05-25:
OK. Thanks.  
$ root dnf install libusbx-devel
...
Installed:
  libusbx-devel-1.0.22-4.fc31.x86_64                                            

Complete!
$ ./configure
...
$ make
...
$

closed 2020-05-25 by rickrich