GPSBabel/gpsbabel #575: feature request: download link to old binary

supervisitor, 2020-05-29:
For the actual version there are download links on web-site to get a binary version directly. But where are the links to a previous binary version? Would be nice to be able to reinstall the version I was using before... and no, the actual version is not working for me, because 1.6 is not working with my OS.

gpsbabel, 2020-05-29:
Sorry. We do not have the resources to support versions indefinitely. We support several years worth of OS and hardware combinations with every release, but we can't do it forever. 

We are open source, so you can always grab the. Source and maintain your own builds.

closed 2020-05-29 by gpsbabel

supervisitor, 2020-06-02:
The work that needs to be done to support it is to do nothing! So not deleting the files and leaving them as they are.
But if you want, you can always find reasons for your actions. Even if such actions often cause more head shaking than understanding. 😉 I am not the only one who has always wondered why the binaries are hidden behind this plan9.php script (search the internet!) instead of being stored here at GIThub, like the "Continuous builds". But now I know...there are still source codes...  (And yes, I know you have to offer some actual links when you want to offer the old binarys... this can be hard work, if you don't know how. 😇)

But today I got good news, someone of the community send me a link where I can download it. The original file with version I was searching for. Cheers!

ping @gpsbabel @robertlipe 

`This is a no offense meant private statement!`