GPSBabel/gpsbabel #597: gpx to csv conversion with more time precision

goamberg, 2020-06-18:
Hi, quick question:

I have some interpolated high precision gpx files and want them to convert to csv. 
My problem is now, that the gpx files have 5 digits behind seconds and only 3 are converted over to csv.

Is there a command like "prec" for lat/lon to define a higher precision in time?

Also interested if this would impact the speed calculations?
Because of those missing digits, the speeds are quite eratic.

Goam

tsteven4, 2020-06-18:
Latitude and longitude in the gpx file should be in Decimal degrees, 
WGS84 datum. We use all the digits to convert to a double.  We use the 
doubles to compute speed.  It is very likely these doubles are stored in 
64 bit IEEE 754 double precision values.

Time in the gpx file should be an ISO 8601 specification for date/time 
representation. Fractional seconds are allowed for millisecond timing in 
tracklogs.  We store time to the millisecond.  This millisecond 
precision time is used to compute speed (for gpx).

The gpsbabel speed calculation does not throw away digits before the 
calculation.  Any truncation is on output.  Different formats print 
differing number of digits.

We have many csv formats.  A few formats have options to control the 
number of digits printed.  You need to be more specific in your use 
case, i.e. what command line are you using to do the conversion?

goamberg, 2020-06-18:
Thanks for the quick reply
Ok got it. With my gpx files I am in the microseconds and therefore out of this ISO spec.

A simple copy move from gpx to unicsv I guess is not possible? 
It will always cut down to milliseconds?

tsteven4, 2020-06-18:
The resolution of our internal concept of time is 1 millisecond. Between 
any reader and any writer the time is stored at this resolution. Some of 
the older formats (not gpx) only support reading and writing to 1 second 
resolution.

goamberg, 2020-06-18:
thanks for the info

closed 2020-06-18 by goamberg

goamberg, 2020-07-21:
Quick question on the precision again:
Is it possible to adjust the precision rounding in height/altitude in unicsv?
Decimetre is a little rough...

robertlipe, 2020-07-22:
Sure. Modify unicsv.cc 's write of wpt->altitude to use a user-settable
option, as is done for llprec.

GPS just isn't that accurate in the Z plane. More numbers doesn't always
mean more precision.

goamberg, 2020-07-22:
thanks for the reply. I looked a little into it and that is way over my head ;)
Personally I dont understand why it shouldn't simply copy the numbers from the gpx over to csv without rounding
I will check ogr2ogr or try to find some python scripts....
Anyway - thanks for taking some time, really appreciated!