GPSBabel/gpsbabel #600: GPX Read error: "Entity 'ntilde' not declared."

Engineer-of-Stuff, 2020-06-24:
I'm merging a lot of gpx files together. I am constantly getting the error `GPX Read error: "Entity 'ntilde' not declared."`

The command I'm using (in this case, with only the file I've attached as a test file. The command is the same with one or more files) is `gpsbabel -i gpx -f downloads/colorado/canon-view-canyon-view.gpx -o gpx -F test.gpx`. The file is not created, the command fails.

[canon-view-canyon-view.zip](https://github.com/gpsbabel/gpsbabel/files/4828241/canon-view-canyon-view.zip)

tsteven4, 2020-06-24:
Your file violates the gpx schema.  This is not a gpsbabel problem.  It is a problem with whatever created canon-view-canyon-view.gpx.
```
Using schemaLocation: "http://www.topografix.com/GPX/1/1 /home/strabert/schema/gpx/gpx11.xsd http://www.garmin.com/xmlschemas/GpxExtensions/v3 /home/strabert/schema/gpx/GpxExtensionsv3.xsd http://www.garmin.com/xmlschemas/TrackPointExtension/v1 /home/strabert/schema/gpx/TrackPointExtensionv1.xsd http://www.topografix.com/GPX/1/0 /home/strabert/schema/gpx/gpx10.xsd"

Fatal Error at file canon-view-canyon-view.gpx, line 3, char 27
  Message: entity 'ntilde' not found

Errors occurred, no output available
```

Engineer-of-Stuff, 2020-06-30:
Ok, but how would you have the letter `Ñ` in a name tag?

tsteven4, 2020-06-30:
The encoding of the file is declared as UTF-8 in the xml declaration of your file.  The UTF-8 encoding for LATIN SMALL LETTER N WITH TILDE is

UTF-8 (hex) | 0xC3 0xB1 (c3b1)

You can use this directly in your file.

[canon-view-canyon-view-fixed.gpx.zip](https://github.com/gpsbabel/gpsbabel/files/4848791/canon-view-canyon-view-fixed.gpx.zip)

robertlipe, 2020-06-30:
Right. GPX is fundamentall XML and not HTML. This input file is (slightly)
broken.

Engineer-of-Stuff, 2020-09-18:
Ok, thanks for the clarification guys.

closed 2020-09-18 by Engineer-of-Stuff