GPSBabel/gpsbabel #605: text.test depends on timezone

tsteven4, 2020-07-12:
This came up in https://salsa.debian.org/debian-gps-team/pkg-gpsbabel/-/jobs/856254 with a different time zone (-1200).

timedatectl set-timezone Pacific/Honolulu
./testo text
```
:~/work/pname$ timedatectl
                      Local time: Sun 2020-07-12 03:50:55 HST
                  Universal time: Sun 2020-07-12 13:50:55 UTC
                        RTC time: Thu 2020-07-09 17:28:57
                       Time zone: Pacific/Honolulu (HST, -1000)
       System clock synchronized: no
systemd-timesyncd.service active: yes
                 RTC in local TZ: no
:~/work/pname$ ./testo text
Running ./testo.d/text.test
--- ./reference/gc/GCGCA8_logs~html.txt	2020-07-04 06:14:38.317635539 -1000
+++ /tmp/gpsbabel.13371/gc_logs.html	2020-07-12 03:50:59.299220452 -1000
@@ -37,28 +37,28 @@
 <img SRC="http://www.mtgc.org/mtgc_member-banner.gif" WIDTH="500" HEIGHT="40" ALT="Member of Middle Tennessee GeoCachers Club [www.mtgc.org]" BORDER="0"></a></p></p>
 <p class="gpsbabelhint"><strong>Hint:</strong> There Is No Hint</p>
 <p class="gpsbabellog">
-<span class="gpsbabellogtype">Found it</span> by <span class="gpsbabellogfinder">littlepod</span> on <span class="gpsbabellogdate">2005-07-03</span><br>
+<span class="gpsbabellogtype">Found it</span> by <span class="gpsbabellogfinder">littlepod</span> on <span class="gpsbabellogdate">2005-07-02</span><br>
 Enjoyed the puzzle. We seemed to be about 50ft off though. TFTC.</p>
 <p class="gpsbabellog">
-<span class="gpsbabellogtype">Write note</span> by <span class="gpsbabellogfinder">robertlipe</span> on <span class="gpsbabellogdate">2005-04-29</span><br>
+<span class="gpsbabellogtype">Write note</span> by <span class="gpsbabellogfinder">robertlipe</span> on <span class="gpsbabellogdate">2005-04-28</span><br>
 TB Drop to show he's hanging out in Nashville until we blast off for Geowoodstock in a few weeks.</p>
 <p class="gpsbabellog">
-<span class="gpsbabellogtype">Found it</span> by <span class="gpsbabellogfinder">Big Bumblebee</span> on <span class="gpsbabellogdate">2005-04-18</span><br>
+<span class="gpsbabellogtype">Found it</span> by <span class="gpsbabellogfinder">Big Bumblebee</span> on <span class="gpsbabellogdate">2005-04-17</span><br>
 Found it a while ago.  Thanks.</p>
 <p class="gpsbabellog">
-<span class="gpsbabellogtype">Write note</span> by <span class="gpsbabellogfinder">robertlipe</span> on <span class="gpsbabellogdate">2005-03-27</span><br>
+<span class="gpsbabellogtype">Write note</span> by <span class="gpsbabellogfinder">robertlipe</span> on <span class="gpsbabellogdate">2005-03-26</span><br>
 I had to renew my permit with the CDC and in doing so, I trolled out here verified that the infectious ooze is fully within specification and industry accepted tolerance.   Ooze On!</p>
 <p class="gpsbabellog">
-<span class="gpsbabellogtype">Found it</span> by <span class="gpsbabellogfinder">Virtual Babe</span> on <span class="gpsbabellogdate">2004-12-27</span><br>
+<span class="gpsbabellogtype">Found it</span> by <span class="gpsbabellogfinder">Virtual Babe</span> on <span class="gpsbabellogdate">2004-12-26</span><br>
 This was a great cache, however on this day I considered it a FIFM cache (Fun, Invigorating, Frustrating and Maddening), especially when the cache was not replaced in the proper spot by the previous cacher!  Thanks anyway!!</p>
 <p class="gpsbabellog">
-<span class="gpsbabellogtype">Write note</span> by <span class="gpsbabellogfinder">robertlipe</span> on <span class="gpsbabellogdate">2004-01-12</span><br>
+<span class="gpsbabellogtype">Write note</span> by <span class="gpsbabellogfinder">robertlipe</span> on <span class="gpsbabellogdate">2004-01-11</span><br>
 I got a complaint from the CDC about oozy rat this weekend.     I went out tonight in the dark and verified that the infectious ooze is fully within specification and industry accepted tolerance. (Although I realize now I did misstate the cache container to the reporting officer when confronted.   It's, uuuuh, smaller than I said.)</p>
 <p class="gpsbabellog">
-<span class="gpsbabellogtype">Write note</span> by <span class="gpsbabellogfinder">robertlipe</span> on <span class="gpsbabellogdate">2003-10-04</span><br>
+<span class="gpsbabellogtype">Write note</span> by <span class="gpsbabellogfinder">robertlipe</span> on <span class="gpsbabellogdate">2003-10-03</span><br>
 In the expectation that this cache will get some traffic in the next 48 hours, Ryan and I checked it earlier today.   The Rat is Oozing just as we planned it.</p>
 <p class="gpsbabellog">
-<span class="gpsbabellogtype">Write note</span> by <span class="gpsbabellogfinder">robertlipe</span> on <span class="gpsbabellogdate">2003-07-03</span><br>
+<span class="gpsbabellogtype">Write note</span> by <span class="gpsbabellogfinder">robertlipe</span> on <span class="gpsbabellogdate">2003-07-02</span><br>
 It won't earn him a smiley face, but I've confirmed that rickrich would have indeed sunk the battleship!      Thanx for playing. You get a copy of the home game and some rice-a-roni...</p>
 </td></tr></table>
 </body></html>
\ No newline at end of file
ERROR comparing ./reference/gc/GCGCA8_logs~html.txt /tmp/gpsbabel.13371/gc_logs.html
--- ./reference/gc/GC7FA4.html	2020-07-04 06:14:38.317635539 -1000
+++ /tmp/gpsbabel.13371/GC7FA4.html	2020-07-12 03:50:59.315228452 -1000
@@ -56,7 +56,7 @@
 PICTURES of points and of the panels will follow soon. YOU CAN ONLY LOG ONE POINT (ONE POINT PER GEOCACHER)
 Good luck!</p>
 <p class="gpsbabellog">
-<span class="gpsbabellogtype">Found it</span> by <span class="gpsbabellogfinder">Christopher R &amp; Pooh B</span> on <span class="gpsbabellogdate">2005-07-12</span><br>
+<span class="gpsbabellogtype">Found it</span> by <span class="gpsbabellogfinder">Christopher R &amp; Pooh B</span> on <span class="gpsbabellogdate">2005-07-11</span><br>
 <span class="gpsbabellogcoords">N48&deg;10.207 W53&deg;58.097</span><br>
 This marker is not in Quebec but it is a Geodesic marker in Clarenville, Newfoundland, Canada!
 
@@ -70,7 +70,7 @@
 
 A trouvé celui-ci tandis que chasse une cachette traditionnelle et pensé à cette cachette tout de suite!  Elle est située sur la montagne nue dans Clarenville - il y a aactually deux marqueurs à moins de 15 pieds d'un des autres sur la montagne nue...  Ours De Pooh De Sourires</p>
[... 52 lines cut ...]
 Près de la cache:  Exo-07 La Jumelle de Loudiver (GCP3VE)
 
-Found it by etasse on 2005-06-03
+Found it by etasse on 2005-06-02
 N45 29.525 W75 43.005
 MRN marker 94K4731 in Gatineau, QC. corner of Du Rhone and Gatineau Ave.
 
@@ -62,11 +62,11 @@
 
 Judging by the coordinates it looks like the coords should be 45°29'31.5" -75°43'0"  I placed the GPS antenna right against the marker, to no avail.
 
-Found it by Katou on 2005-06-03
+Found it by Katou on 2005-06-02
 N46 37.091 W71 55.974
 Un bo point géodésique a Lotbinière..en allant faire une nouvelle cache a l'île richelieu ;-)
 
-Found it by Gps_Gulliver&DauphinBleu on 2005-05-29
+Found it by Gps_Gulliver&DauphinBleu on 2005-05-28
 N45 32.449 W73 30.747
 Point Geodesique situe near Port de Plaisance de Longueuil
 sur le bord du fleuve st-laurent.
ERROR comparing ./reference/gc/GC7FA4.text /tmp/gpsbabel.13371/GC7FA4.text
Running UTF-8 BOM test
Running well-formed XML test
```

tsteven4, 2020-07-12:
It is unclear what timezone <groundspeak:date> elements are using.
In http://static.groundspeak.com/cache/1/0/cache.xsd and http://static.groundspeak.com/cache/1/0/1/cache.xsd they are a type xs:string
However, in http://static.groundspeak.com/cache/1/2/cache.xsd they are type xs:dateTime.  

xs:dateTime without a time zone value represents and unknown time zone, it is not UTC.

There is a history of difficulty with groundspeak dates and timezones.  It is possible groundspeak:date elements are always in Seattle time.

tsteven4, 2020-07-12:
After 1.6.0 text.test was enhanced to increase coverage.  The detection of the failure in 1.7.0 is due to this increased coverage rather than a change in behavior.

robertlipe, 2020-07-13:
That's a ghost from the past.

Groundspeak has long had problems in this area. When these snaps were
taken, Grounspeak was using Groundpeak Time Zone for *almost* everything in
a pocket query. The other reality is that the hours almost never matter,
except when they cause rounding of dates which tends to happen a lot as
people tend to find - and long - late in the day.

Text.cc has one of our few occurrences of 'foo = localtime()' which may be
some of our pain. Bouncing logtime through a time_t and messing with time
zones at all is probably a bad idea. If we made logtime a QDate and nailed
the TZ to GMT (or Pacific something Time) would we be better off?

Do any of you geoachers here know if Groundspeak ever fixed this and moved
the TZ to GMT or such? Would just pulling fresh versions of our 15 y/o
samples and regenerating references make some of this go away?

RJL