GPSBabel/gpsbabel #625: Check for upgrade fails

tsteven4, 2020-08-11:
Windows 10, gpsbabel 1.7.0, Check for upgrade yields:

[image]

tsteven4, 2020-08-11:
related to #580, #587.

tsteven4, 2020-08-11:
Running a debug build on windows with testing set I see
`redirect to  "https://www.gpsbabel.org/upgrade_check.html"`

@gpsbabel is www.gpsbabel.org doing a redirect on upgrade_check.html now?  Without the specific version of openssl installed on windows a https request will fail.

robertlipe, 2020-08-11:
We've ping-ponged on this. I can look up the history/details if needed, but
let's think about what's in the field now and where we want to go.

There was a big push a few years ago to move the entire web to https.
Google search and webmaster tools pretty much insisted on it. Let's Encrypt
made it pretty easy.
As of this moment, all https requests are rewritten and served via a
redirect.

% curl http://www.gpsbabel.org/upgrade_check.html
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>*301 Moved Permanently*</title>
</head><body>
<h1>Moved Permanently</h1>
<p>The document has moved <a href="
https://www.gpsbabel.org/upgrade_check.html">here</a>.</p>
</body></html>

At one point, we special cased that one URL, but removed that special case
because of something like this that broke in the field.

Part of our .htaccess includes:
# RewriteCond %{REQUEST_URI} !^/upgrade_check.html$

Why is QtNetwork requiring a version of a library and not relying
on/providing a working one? Did we just forget to package a .lib or
something?

Upgrade_check is actually a program and not a static file. We could try to
do something Windows-specific when servicing that POST but that seems
disturbing. We don't really get a meaningful UA, but we could probably
special case Windows if we really really had to. Let's not.

Sidebar: access_log is showing *way* fewer hits on upgrade_check than I'd
expect. Even rate-limited, I'd expect more than the tree copies we served
on Aug 6, for example.

RJL

robertlipe, 2020-08-11:
Oh. those numbers were the paltry http ones that were redirected. The
"real" ones fetched over https are over a thousand a day, which is more
what I'd expect.

So, back to the original question: we currently ARE redirecting and without
the public outcry I'd expect if we splashed an error for every Windows
user. What should we do and how can we be sure we don't leave them orphaned
when 1.7.1 (whatever) hits?

RJL

tsteven4, 2020-08-11:
It's probably because Qt changed.  1.6.0 on Windows was 5.12.2.  1.7.0 was 5.12.6.  Qt went to open ssl 1.1.1 with TLS 1.3 in 5.2.4.  It caused quite an uproar.

In https://github.com/gpsbabel/gpsbabel/pull/587 on June 2, 2020 you commented referring to the upgrade request:

> There's a special case on the server to NOT redirect that one url to https so I think this is ok.

Perhaps a better solution is to use QDesktop servers and just farm it out to a browser like we do for help,etc. 

Qt lays it out the requirements here:
https://doc.qt.io/qt-5/windows-requirements.html

robertlipe, 2020-08-24:
Unfortunately, I just rediscovered this problem. Again.

The code on the server that was supposed to handle the special case of NOT
redirecting from http to https was failing, so we were flinging back a 301
and this code couldn't handle it.

There is debug code in upgrade.cc that recognizes the redirect, but by this
time we've lost so much context we can't really restart the query.

I've added code in upgrade.cc that tells it to retry if it ever does see
the redirect, but we have to special case it for the existing user base.

We are now allowing http end-to-end for this lone request.

 857d7414..d83b11f1

RJL