GPSBabel/gpsbabel #627: qmake errors

rickrich, 2020-08-13:
$ qmake
GPSBabel.pro:12: Unknown test function: lessThan
GPSBabel.pro:12: Unknown test function: if
GPSBabel.pro:12: Unknown test function: if
GPSBabel.pro:40: Unknown test function: defined
GPSBabel.pro:45: Unknown test function: defined
Project LOAD(): Feature configure cannot be found.

Fedora 31.

tsteven4, 2020-08-13:
Fedora renames Qt 5's qmake to the non-standard name of qmake-qt5.

rickrich, 2020-08-13:
You should indicate the process in INSTALL.

rickrich, 2020-08-13:
$ ll /usr/bin/qmake-*
-rwxr-xr-x. 2 root root 4386920 Jul 28  2019 /usr/bin/qmake-qt4
-rwxr-xr-x. 2 root root 3010136 Apr  8 06:01 /usr/bin/qmake-qt5

I guess there was a problem with combining them.  Sorta like Python and two versions.

tsteven4, 2020-08-13:
With Fedora 32 I don't find qmake at all.  If I use qmake-qt4 I get a pretty fair message:

qmake-qt4

Project ERROR: /usr/bin/qmake-qt4 uses Qt version 4.8.7 but version 5.9.0 or newer is required.

rickrich, 2020-08-13:
$ ll /usr/lib64/qt-3.3/bin/qmake
-rwxr-xr-x. 1 root root 1829416 Jul 26  2019 /usr/lib64/qt-3.3/bin/qmake

rickrich, 2020-08-13:
$ root dnf remove qt3.x86_64   qt3-devel.x86_64
Dependencies resolved.
================================================================================
 Package                  Arch   Version                    Repository     Size
================================================================================
Removing:
 qt3                      x86_64 3.3.8b-79.fc31             @fedora        13 M
 qt3-devel                x86_64 3.3.8b-79.fc31             @fedora        43 M
Removing dependent packages:
 google-earth-stable      x86_64 7.1.8.3036-0               @google-earth 180 M
 google-musicmanager-beta x86_64 1.0.467.4929-0             @@commandline  16 M
Is this ok [y/N]: n
Operation aborted.
$

Sorry, both of those (google-earth, google-musicmanage) I use.

robertlipe, 2020-08-13:
Qt3. 3 was obsolete about 15 years ago. It's probably too old to even spit
up a good error.

I agree that we need to document these things as we learn them. We simply
don't have everything in qi and the vendors aren't helping much. The
distros seem intent on shipping "creative" builds.

If you build work qt5, which we do document as required, are we good?

robertlipe, 2020-08-13:
Rick, something is weird.

I was personally the one the moved Earth off Qt3 in 2007, though it may not
have shipped until 2008. This was back around Earth 4.2 or 4.3 and that's
so old it won't even render tiles served by the imagery pipeline.  Any
Earth so old to use Qt3 would display an error on startup.

Is it possible you have a damaged package database r something? Maybe you
have a version modified by someone that has some tool included that relies
on Qt3, but Earth itself really can't be depending on Qt3 these days.

Debugging Earth's packaging is a distraction, though. I don't think
GPSBabel ever required Qt3 and we've been on Qt5 for years now. I know
you've been building since then. What's changed?

tsteven4, 2020-08-14:
Robert,
What changed is configure was deleted so he moved to qmake.  I am not sure that qmake is even in /usr/bin on Fedora 31, I think he may have found in /usr/lib64/qt3-3/bin and invoked it from there.  In what I regard as a mistake, long ago, I taught configure to check for qmake-qt5, which is an aberration of Fedora/RHEL/Centos and their derivatives.  My view now is that is up to the user to invoke Qt5's qmake in whatever fashion is appropriate on that distribution.  From there we assume responsibility to invoke other Qt tools, e.g. lupdate, lrelease, based on that version of qmake.

rickrich, 2020-08-14:
qmake is set by my PATH via

$  root  dnf install qt3

 and /etc/profile.d/qt.sh.  I did not change this. 

```
$ cat /etc/profile.d/qt.sh
# Qt initialization script (sh)

# In multilib environments there is a preferred architecture, 64 bit over 32 bit in x86_64,
# ppc64. When a conflict is found between two packages corresponding with different arches,
# the installed file is the one from the preferred arch. This is very common for executables
# in /usr/bin, for example. If the file /usr/bin/foo is found  in an x86_64 package and in
# an i386 package, the executable from x86_64 will be installe

if [ -z "${QTDIR}" ]; then

case `uname -m` in
   x86_64 | ia64 | s390x | ppc64 | ppc64le)
      QT_PREFIXES="/usr/lib64/qt-3.3 /usr/lib/qt-3.3" ;;
   * )
      QT_PREFIXES="/usr/lib/qt-3.3 /usr/lib64/qt-3.3" ;;
esac

for QTDIR in ${QT_PREFIXES} ; do
  test -d "${QTDIR}" && break
done
unset QT_PREFIXES

case :$PATH: in
    *:$QTDIR/bin:*) ;;
    *) PATH=$QTDIR/bin:$PATH ;;
esac

QTINC="$QTDIR/include"
QTLIB="$QTDIR/lib"

export QTDIR QTINC QTLIB PATH

fi
$ 
```

tsteven4, 2020-08-14:
@rickrich As I said 6 hours ago, because fedora has chosen to rename things, you need to use qmake-qt5 to build gpsbabel.  Unless this is a new system you already had that installed because that is what configure would have looked for.  I think the fedora package is qt5-qtbase-devel.

Also, you are using old version of google earth and google music manager.  They have newer ones available based on Qt 5.  This really has nothing to do with gpsbabel, Fedora is still going name the Qt5 qmake program qmake-qt5.

rickrich, 2020-08-14:
```
$ root dnf install google-musicmanager-beta_current_x86_64.rpm
Last metadata expiration check: 1:09:03 ago on Thu 13 Aug 2020 10:22:09 PM CDT.
Dependencies resolved.
================================================================================
 Package                     Arch      Version            Repository       Size
================================================================================
Installing:
 google-musicmanager-beta    x86_64    1.0.467.4929-0     @commandline    5.2 M
Installing dependencies:
 qt3                         x86_64    3.3.8b-79.fc31     fedora          3.7 M
 redhat-lsb                  x86_64    4.1-48.fc31        fedora           18 k
 redhat-lsb-desktop          x86_64    4.1-48.fc31        fedora           14 k

Transaction Summary
================================================================================
Install  4 Packages

Total size: 9.0 M
Total download size: 3.7 M
Installed size: 29 M
Is this ok [y/N]: n
Operation aborted.
```

Where is the new google-musicmanager rpm?

tsteven4, 2020-08-14:
Thats the one I found.  It directly depends on qt5, but somewhere in the tree it pulls in qt3 as well

tsteven4, 2020-08-14:
despite the appearance of support for alternatives in https://src.fedoraproject.org/rpms/qt5-qtbase/blob/master/f/qt5-qtbase.spec
it doesn't seem to work.

1. You may be able to create a link yourself and then use the qmake command:
sudo alternatives --install /usr/bin/qmake qt /usr/lib64/qt5/bin/qmake 100
2. or just use qmake-qt5 
3. or put /usr/lib64/qt5/bin early enough in your PATH.

tsteven4, 2020-08-14:
#630 added CI against Fedora 32.  See tools/Dockerfile_f32 and .travis.yml.  This uses the alternative method mentioned above to overcome the naming difficulties with qmake.  It serves as an example on how builds can be done on Fedora.

closed 2020-08-14 by tsteven4