GPSBabel/gpsbabel #637: xmlstreamwriter use of codec

tsteven4, 2020-08-25:
In Qt6 QXmlStreamWriter always writes utf-8.  In our xmlstreamwriter we create a codec to filter out some characters which is no longer supported.  It looks like this was fixed in Qt in https://github.com/qt/qtbase/commit/3b5b8f1d4ab8092e5dd337b7b4e32d85fda2e0b7 likely making our fix unnecessary.  The Qt fix will set hasError.  The fix appeared in 5.11.0, so it isn't in our floor of 5.9.

GPSBabelDeveloper, 2021-01-26:
5.9 was an LTS release and its 3 year window expired on 31 May 2020. 5.12, the next LTS, expires in Dec 2021.

With 5.15 (May 2020) being a de-facto LTS and it bringing in many shims and marked deprecations that will help us when Qt6 is ready for us, I wonder if we should just move to (cringe) 5.15. We dodged several bullets with deprecations and you've already fixed the ones I found by inspection. (Thank you.)

IMO, we should make the move to at least 5.12. We've missed 2/3rd of its life cycle.

We just don't have a significant pool of developers to cater to these days.  The increased stability in the geo world over 15 years ago plus the widespread use of containers to let you separate a runtime from what the vendor ships means that people just don't _need_ to build a development GPSBabel with a 3-5 year old toolchain like they once did.

Is it time to go to 5.15.3 as a stepping stone to 6.someday ?

tsteven4, 2021-01-26:
As usual we are held back by Linux.  ubuntu bionic has Qt 5.9.5 and has standard support until April 2023.  For development ubuntu focal is now available with Qt 5.12.8.  A bunch of our CI still is on bionic, but it could be moved up to focal.    This would preclude distributions like bionic from upgrading gpsbabel, but it likely they ever will (bionic is still at 1.5.4, focal is at 1.6.0, groovy is at 1.7.0).

We didn't miss 2/3 of the 5.12 life cycle.  Our 1.6.0 and 1.7.0 releases for windows and macOS both included 5.12.x.

When 5.15.3 is released I think it will be commercial only.  I think the source will NOT be publicly available in git or as a tarball.  I think 5.15.2 is the last release of 5.15 that will have source that is publicly available.

I agree there are a bunch of compatibility breaking changes between Qt 5.9 and Qt 6.0.  Many of these we could get a head start on with Qt 5.15.2.  But I am in no hurry as 6.0 doesn't work for us, and I believe they said our codec issue may or may not be resolved in 6.2, but not before then.

Moving the floor to 5.12 allows us to fix this xmlstreamwriter issue, but I am not aware of anything else it enables.

I'll support moving the floor to 5.12, but I don't think it buys us anything besides the xmlstreamwriter issue.

PR #671 (2021-01-27): trust Qt to strip out characters illegal in xml.
this feature was added in Qt 5.11.0.

this resolves #637
closed 2021-01-27 by tsteven4