GPSBabel/gpsbabel #644: Problem with import to MyTourbook

Starmbi, 2020-10-08:
Hello,

after i simplified a gpx-File with gpsbabel, i was not able to import this file to MyTourbook.
So i created a issue at the Github page from MyTourbook [https://github.com/wolfgang-ch/mytourbook/issues/237#issuecomment-705658740](https://github.com/wolfgang-ch/mytourbook/issues/237#issuecomment-705658740)

Here a part of a answer:

> My hunch is that the culprit is the topografix namespace added by GPS Babel that is wrong :
> http://www.topografix.com/GPX/ 8/2
> 
> This is the value that I would expect knowing the GPS schema : http://www.topografix.com/GPX/1/1
> 
I work with the newest version 1.7.0 for windows.

Thanks in advance

Stefan

tsteven4, 2020-10-08:
The gpx element version attribute is certainly wrong in https://github.com/wolfgang-ch/mytourbook/files/5349153/simple_5m.gpx.txt

It is possible create an invalid file like this by using the gpxver option with gpsbabel.  Did you do something like
./gpsbabel -i gpx -f reference/bounds-test.gpx -o gpx,**gpxver=8.2** -F -

Starmbi, 2020-10-08:
Hello,

i downloaded the gpx from Kommot.de and and tried some different Apps to work with the gpx.
After that i got:
```
<?xml version="1.0" encoding="UTF-8"?>
<gpx xmlns="http://www.topografix.com/GPX/1/1" xmlns:gpxdata="http://www.cluetrust.com/XML/GPXDATA/1/0" creator="Komoot" version="8.2">
```
As you can see, there is a **version 8.2**.
Now i downloaded the original file from Komoot again and found:
```
<?xml version='1.0' encoding='UTF-8'?>
<gpx version="1.1" creator="https://www.komoot.de" xmlns="http://www.topografix.com/GPX/1/1" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://www.topografix.com/GPX/1/1 http://www.topografix.com/GPX/1/1/gpx.xsd
```
I think, one of the Apps, I tested, wrote this version into the header.
Now the converted file with gpsbabel looks like:
```
<?xml version="1.0" encoding="UTF-8"?>
<gpx version="1.1" creator="GPSBabel - https://www.gpsbabel.org" xmlns="http://www.topografix.com/GPX/1/1" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"> 

```
So, all is OK, it works and this issue can be closed.

Sorry
Stefan

gpsbabel, 2020-10-08:
User never said where his or her 8.2 came from, but it certainly want us. 

As asked, closing.

closed 2020-10-08 by gpsbabel