GPSBabel/gpsbabel #645: Question:  Ozi codec

velocat, 2020-10-19:
Installed Gpsbabel in Debian, stable version.
-V = 1.5.7
From which version is the codec argument enabled for the Ozi format ?
It doesn't work for me.
And how to solve this problem?

tsteven4, 2020-10-19:
The codec option in the ozi format was introduced in gpsbabel version 
1.6.0, so you will need 1.6.0 or 1.7.0.

velocat, 2020-10-19:
And when are these versions planned to be stable in the Debian repository?

tsteven4, 2020-10-19:
That's a question for Debian, not us.  You can clone this repo or grab a tarball from http://www.gpsbabel.org/download.html and build from source.

velocat, 2020-10-22:
Decided. Build and install version 1.7.0

closed 2020-10-22 by velocat

robertlipe, 2020-10-29:
Be sure and thank them for disabling our built-in upgrade check, too.