GPSBabel/gpsbabel #646: Question by filtering

velocat, 2020-10-24:
Tell me, I can't discard part of the data for the output file if it's not gpx.
For example, I want to filter data from a KML file to the same format
, but I get absolutely all the data in the output.
-w -t -r - not effect
nuketypes - not effect
```
gpsbabel -D9 -w -i gpx -f 'file1.gpx' -x nuketypes,tracks,routes -i kml -f '.file2.kml' -x nuketypes,tracks,routes -o kml,unit=m,points=0 -F /tmp/temp.kml
```

I see as an option only save the file as gpx (filtered), and the second pass to save it in the desired format. But this is unnecessary, I think. :(

Maybe I'm doing something wrong?

tsteven4, 2020-10-24:
Perhaps you meant to write .file2.kml?  If so you need to use -F instead 
of -f.

To monitor waypoints, routes, tracks you can add -x validate,debug.

The following shows bounds-test.gpx has 11 waypoints and three routes 
before the nuketypes filter.   After the nuketypes filter only 11 
waypoints remain.

> ./gpsbabel -i gpx -f reference/bounds-test.gpx -x validate,debug -x 
> nuketypes,routes,tracks -x validate,debug -o gpx -F out.gpx
>
> Processing waypts
> point ct: 11, waypt_count: 11
>
> Processing routes
> route 0 ct: 60, waypt_count: 60
> route 0 ct: 35, waypt_count: 35
> route 0 ct: 13, waypt_count: 13
> route head ct: 3, route_count: 3
> total route point ct: 108, route_waypt_count: 108
>
> Processing tracks
> track head ct: 0, track_count: 0
> total track point ct: 0, track_waypt_count: 0
>
> Processing waypts
> point ct: 11, waypt_count: 11
>
> Processing routes
> route head ct: 0, route_count: 0
> total route point ct: 0, route_waypt_count: 0
>
> Processing tracks
> track head ct: 0, track_count: 0
> total track point ct: 0, track_waypt_count: 0

On 10/24/2020 12:42 AM, velocat wrote:
> |gpsbabel -D9 -w -i gpx -f 'file1.gpx' -x nuketypes,tracks,routes -i 
> kml -f '.file2.kml' -x nuketypes,tracks,routes -o kml,unit=m,points=0 
> -F /tmp/temp.kml|

velocat, 2020-10-24:
Well, I wrote that there are no problems with filtering the gpx file.
But how can I filter out another format, such as xml?

my example shows 2 gpx and kml files (-f), from which I want to get the third one (-F), but I want to discard tracks and routes in it and leave only waypoints

tsteven4, 2020-10-24:
The filters don't know or care where the waypoints/routes/tracks came from.

Try this:

gpsbabel -i gpx -f 'file1.gpx' -i kml -f 'file2.kml' -x validate,debug -x nuketypes,tracks,routes -x validate,debug -o kml,unit=m,points=0 -F /tmp/temp.kml

The validate output should show that after the nuketypes filter any routes and tracks are gone.

For more help be sure to give your exact command line and the relevant files, i.e.
attach file1.gpx file2.kml temp.kml

velocat, 2020-10-27:
Thanks, with the help of "-x validate,debug" I found my mistake: using the console I wrote commands correctly, but calling from scripts, I missed one letter in nuketypes - instead of track**s** I wrote track))

But I have another question about this:
I am using GpsBabel from PHP by calling it with the command
```
$cmd = 'gpsbabel -i... -f.... etc';
exec($cmd, $output, $return_var);
```
I would like to catch errors in the script, but I don't really understand how to do it.
Suppose I have a file with no waypoints. Then by running:
```
$cmd = 'gpsbabel -w -i gpx -f "2019.gpx" -x nuketypes,tracks,routes -x validate,debug -o ozi,altunit=m,codec=utf-8 -F /tmp/temp_YbEh1J.wpt';
exec($cmd, $output, $return_var);
```
i get the following:
```
return_var = 139
output = array(0) { } 
```
If executed in the console, then I see the following:
```
Processing waypts
point ct: 0, waypt_count: 0

Processing routes
route head ct: 0, route_count: 0
total route point ct: 0, route_waypt_count: 0

Processing tracks
track head ct: 0, track_count: 0
total track point ct: 0, track_waypt_count: 0
Segmentation fault
```
I would like to learn how to handle messages and errors in my script (php). Can you please tell me how this can be done? Now I can only check whether the command completed successfully or not.

gpsbabel, 2020-10-27:
On Tue, Oct 27, 2020 at 12:40 AM velocat <notifications@github.com> wrote:

Thanks, with the help of "-x validate,debug" I found my mistake: using the
> console I wrote commands correctly, but calling from scripts, I missed one
> letter in nuketypes - instead of track*s* I wrote track))

Bummer.

I just confirmed we do issue a warning for that. You're right that your
program needs to be watching for warnings on stderr.

% gpsbabel -wrt -i random,points=50  -f /dev/null -x
nuketypes,track,nonsense -o gpx -F /dev/null
'track' is an unknown option to nuketypes.
'nonsense' is an unknown option to nuketypes.

I am using GpsBabel from PHP by calling it with the command
>
> $cmd = 'gpsbabel -i... -f.... etc';
> exec($cmd, $output, $return_var);
>
> I would like to catch errors in the script, but I don't really understand
> how to do it.
>
We're really not a PHP support forum, but it looks like you want proc_open
so you can capture stderr (which is getting the 'debug'  chatter from the
validate filter) on a separate pipe than stdout. With those keywords as a
hint, you may start with, say:

https://stackoverflow.com/questions/2320608/php-stderr-after-exec

It's the equivalent in common shells stdout being the default filescript of
1 and stderr on 2 needing special handling. That's the "2" in the
redirection in this shell fragment:

 % gpsbabel -wrt -i random,points=50  -f /dev/null -x
nuketypes,track,nonsense -o gpx -F /dev/null 2> /tmp/warnings
 % cat /tmp/warnings
'track' is an unknown option to nuketypes.
'nonsense' is an unknown option to nuketypes.

> $cmd = 'gpsbabel -w -i gpx -f "2019.gpx" -x nuketypes,tracks,routes -x validate,debug -o ozi,altunit=m,codec=utf-8 -F /tmp/temp_YbEh1J.wpt';
> exec($cmd, $output, $return_var);
>
>   [ ... ]
>
> Processing tracks
> track head ct: 0, track_count: 0
> total track point ct: 0, track_waypt_count: 0
> Segmentation fault
>
>
Crashes are bad. If this is reproducible with our development head or out
most recent release, please provide a full test case so we can investigate
it.

We should be able to recover/error from even hideously malformed GPX input.
We control what we process in the Ozi writer so if there is something we're
just misunderstanding, I'd like to at least see a file and command line
invocation  that sends us into the guardrails.

RJL

velocat, 2020-10-27:
Here's an example: (GPSBabel Version 1.7.0)
```
gpsbabel -w -i gpx -f  "11_kavkaz 2019.gpx" -x nuketypes,tracks,routes -x validate,debug -o ozi,altunit=m,codec=utf-8 -F /tmp/temp_y6byrz.wpt
```
This happens if suddenly there are no waypoints in the track:
```
Processing waypts
point ct: 0, waypt_count: 0

Processing routes
route head ct: 0, route_count: 0
total route point ct: 0, route_waypt_count: 0

Processing tracks
track head ct: 0, track_count: 0
total track point ct: 0, track_waypt_count: 0
Segmentation fault
```
input file: https://yadi.sk/d/W202DTTLgS2r6A

robertlipe, 2020-10-27:
Perfect. If we can catch it in a debugger, making it not crash is easy.

https://github.com/gpsbabel/gpsbabel/pull/647 is running regression tests
now. I'm not totally satisfied with that, but it's a safe prescription.

Thanx,
RJL

velocat, 2020-10-31:
In order not to open a new question, I will ask here. ))
1. Let's take an example from the description:
`gpsbabel -t -i gpx -f in.gpx -x track,split,title="ACTIVE LOG # %Y%m%d" -o gpx -F out.gpx`

I want to split the track by day. But the track is recorded + 3UTC, and it is divided by a timestamp three hours less, i.e. in my case, the separation occurs at 21 o'clock. Tell me how to make a correction?

It seems to me that it would be strange to "move" the time of the entire track, because then it will display three hours more than it actually does.

2. Using split, I end up with several tracks, but the first of them has an extension attached that contains data for the entire original track.
```
....
 <trk>
    <name>tst \#15-10-2020</name>
    <extensions>
      <gpxx:TrackExtension>
        <gpxx:DisplayColor>Red</gpxx:DisplayColor>
      </gpxx:TrackExtension>
      <gpxtrkx:TrackStatsExtension xmlns:gpxtrkx="http://www.garmin.com/xmlschemas/TrackStatsExtension/v1">
        <gpxtrkx:Distance>39606.030</gpxtrkx:Distance>
        <gpxtrkx:TimerTime>11893.703</gpxtrkx:TimerTime>
        <gpxtrkx:MinElevation>91.020</gpxtrkx:MinElevation>
        <gpxtrkx:MaxElevation>137.220</gpxtrkx:MaxElevation>
        <gpxtrkx:Ascent>300.120</gpxtrkx:Ascent>
        <gpxtrkx:Descent>-255.180</gpxtrkx:Descent>
      </gpxtrkx:TrackStatsExtension>
    </extensions>
    <trkseg>
......
    </trkseg>
 </trk>
  <trk>
    <name>tst \#16-10-2020</name>
    <trkseg>
......
    </trkseg>
 </trk>
.....
```

Are there any options to use (recalculate) these extensions for all created new tracks, or at least remove this information from the first one, since it is not correct for him?

tsteven4, 2020-10-31:
1. The splits should occur by days in local time irregardless of how the track is recorded.  If this isn't the behavior you observe please provide a test case.
2. The extension data might be being passed through, which is something our gpx format does when it encounters elements it doesn't understand.  Providing a test case would make it easier to analyze this issue.

velocat, 2020-10-31:
Where do the values ​​for Split come from? From points or track metadata?
I keep trying different options, but I noticed that separation does not occur at all if I apply the filter```gpsbabel -D1 -r -t -i gpx -f D:/GARMIN/2020-10-26 21_30_14.gpx -x nuketypes,waypoints -x track,split=1h,title=LOG #%c -o gpx -F D:/GARMIN/tstNew.gpx

options: module/option=value: gpx/snlen="32" (=default)
options: module/option=value: gpx/elevprec="3" (=default)
options: module/option=value: nuketypes/waypoints="1"
options: module/option=value: nuketypes/tracks="0" (=default)
options: module/option=value: nuketypes/routes="0" (=default)
options: module/option=value: track/split="1h"
options: module/option=value: track/merge=""
options: module/option=value: track/title="LOG #%c"
options: module/option=value: track/course="1"
options: module/option=value: track/speed="1"
options: module/option=value: gpx/snlen="32" (=default)
options: module/option=value: gpx/elevprec="3" (=default)
```
In: https://yadi.sk/d/Bn_WA8vPYM3bbQ
Out: https://yadi.sk/d/NqqSb7VesTuoDA
what am I doing wrong?

velocat, 2020-10-31:
> 1. The splits should occur by days in local time irregardless of how the track is recorded.  If this isn't the behavior you observe please provide a test case.

In fact, Z is set in timestamps, so this question is not relevant, everything is happening correctly, I just did not take into account that the input data is at zero UTC

tsteven4, 2020-10-31:
The time values for split come from each point.
If you supply a value for the split option, e.g. split=1h, then instead of splitting by days it will split whenever the time between successive points exceeds the given value.  See https://www.gpsbabel.org/htmldoc-development/filter_track.html:
`Additionally you can add an interval to the split option. With this the track will be split if the time between two points is greater than this parameter. The interval must be numeric and can be int days, hours, minutes or seconds, expressed as one of the character "d", "h", "m", or "s". `

Note that the gpx schema says times are always recorded in UTC.

velocat, 2020-10-31:
I realized my mistake in understanding)))
Exceeds interval is the key expression. And I just wanted to split the track by the hour.
That is, create tracks for each hour.
Apparently this is impossible?

tsteven4, 2020-10-31:
Currently it is not possible to split by hours.  There is an indication in the code that possibly anticipated the need for something like this, but it has not been implemented.  You could of course hack up the code for your own use to make it split by hours.

https://github.com/gpsbabel/gpsbabel/blob/e1e195beca8e0f24ca0e41c02d1986af13865cd0/trackfilter.cc#L504-L507

velocat, 2020-10-31:
@tsteven4  I have a build on Debian, so it's not a good idea for me to change the code and rebuild, so I'll have to forget about this possibility. Thanks.

> 2\. The extension data might be being passed through, which is something our gpx format does when it encounters elements it doesn't understand.  Providing a test case would make it easier to analyze this issue.

But on the issue of expanding tracks, I am attaching the output file, which turned out when dividing for example by a minute.
As a result, I get several tracks, and only the first one has an extension. Moreover, this is an extension of the original track, and not this part of the separated ones.

It would be logical to either have extensions with up-to-date data for each new track, or remove this extension from the first track altogether, since it is not correct for it.

In: in the post above
Out: https://yadi.sk/d/bGoCgFWFzLeE3Q

tsteven4, 2020-10-31:
One way to work around this is to first translate to a format that doesn't support the extension data, and then split it.  You'll need to invoke gpsbabel twice to have this work.
gpsbabel -t -i gpx -f  /media/sf_Shared/2020-10-26\ 21_30_14.gpx -o unicsv -F x.csv
gpsbabel -t -i unicsv -f x.csv -x track,split=1m -o gpx -F x.gpx

robertlipe, 2020-10-31:
Remember that reading, filtering, and writing are all different phases for
us. If the read attaches something we don't really understand, we keep it
attached, but it's not possible for those stages to recompute them.

tsteven4, 2020-10-31:
The issue is that the original track has fs data that pertains to the entire original track.  When we run through the track filter to split we create new route heads every time we split, but we still have the original route head with the now inappropriate fs data.  In this case, a one line fix, is to have the track filter destroy the fs chain of the original route head.  While that is a good thing to do in this specific use case it is clear that dropping the route fs data is always desired when the contents of the route change.

```
$ git diff trackfilter.cc
diff --git a/trackfilter.cc b/trackfilter.cc
index 2c3682cc..cf40d3d5 100644
--- a/trackfilter.cc
+++ b/trackfilter.cc
@@ -409,6 +409,7 @@ void TrackFilter::trackfilter_split()
     fatal(MYNAME "-split: Cannot split more than one track, please pack (or merge) before!\n");
   } else if (!track_list.isEmpty()) {
     route_head* master = track_list.first();
+    master->fs.FsChainDestroy();
     int count = master->rte_waypt_ct;
 
     int i, j;
```

velocat, 2020-11-01:
Tell me more, please ))
Continuing my experiments, I did not understand the purpose of such options as speed and course.
Adding them to the filter line, I did not get any speed or course at the output.
Perhaps I don't understand their purpose? I thought they were calculating adding appropriate values to all points on the track...

tsteven4, 2020-11-01:
For general questions you can address a larger audience of knowledgeable users using the general discussion mailing list https://www.gpsbabel.org/lists.html.

Indeed they are, as documented on https://www.gpsbabel.org/htmldoc-development/filter_track.html

$ ./gpsbabel -t -i gpx -f /media/sf_Shared/2020-10-26\ 21_30_14.gpx  -o unicsv -F - | head -n 10
No,Latitude,Longitude,Altitude,Date,Time
1,55.598527,38.122117,139.0,2020/10/26,10:31:39
2,55.598579,38.122064,139.1,2020/10/26,10:31:49
3,55.598537,38.122151,139.2,2020/10/26,10:31:55
4,55.598458,38.122198,139.3,2020/10/26,10:32:00
5,55.598379,38.122214,139.5,2020/10/26,10:32:04
6,55.598374,38.122219,139.5,2020/10/26,10:32:05
7,55.598368,38.122250,139.5,2020/10/26,10:32:07
8,55.598367,38.122284,139.6,2020/10/26,10:32:08
9,55.598366,38.122326,139.6,2020/10/26,10:32:09
$ ./gpsbabel -t -i gpx -f /media/sf_Shared/2020-10-26\ 21_30_14.gpx  -x track,speed,course -o unicsv -F - | head -n 10
No,Latitude,Longitude,Altitude,Speed,Course,Date,Time
1,55.598527,38.122117,139.0,0.00,0.0,2020/10/26,10:31:39
2,55.598579,38.122064,139.1,0.67,330.1,2020/10/26,10:31:49
3,55.598537,38.122151,139.2,1.20,131.0,2020/10/26,10:31:55
4,55.598458,38.122198,139.3,1.86,161.2,2020/10/26,10:32:00
5,55.598379,38.122214,139.5,2.21,173.8,2020/10/26,10:32:04
6,55.598374,38.122219,139.5,0.60,147.0,2020/10/26,10:32:05
7,55.598368,38.122250,139.5,1.04,110.5,2020/10/26,10:32:07
8,55.598367,38.122284,139.6,2.16,93.2,2020/10/26,10:32:08
9,55.598366,38.122326,139.6,2.62,92.0,2020/10/26,10:32:09

velocat, 2020-11-01:
> For general questions you can address a larger audience of knowledgeable users using the general discussion mailing list https://www.gpsbabel.org/lists.html.
Thank you, I tried to use this resource, but it is very confusing for me, and to find any answer you need to spend a lot of time, I first use the search there, and only when I don't find the answer, I ask a question here ))

> $ ./gpsbabel -t -i gpx -f /media/sf_Shared/2020-10-26\ 21_30_14.gpx -x track,speed,course -o unicsv -F - | head -n 10
That is, as I understand it, these options do not work for the gpx output format, but only for unicsv?
Just for me, the gpx format is the main one, as in principle it is said here in the manual ))

tsteven4, 2020-11-01:
I meant feel free to ask a question using the mailing list gpsbabel-misc.

course and speed are only elements in gpx version 1.0, they were removed from gpx version 1.1.  You can force gpx version 1.0, and then course and speed will be included if they exist:

```
$ ./gpsbabel -t -i gpx -f /media/sf_Shared/2020-10-26\ 21_30_14.gpx  -x track,speed,course -o gpx,gpxver=1.0 -F - | head -n 20
<?xml version="1.0" encoding="UTF-8"?>
<gpx version="1.0" creator="GPSBabel - https://www.gpsbabel.org" xmlns="http://www.topografix.com/GPX/1/0" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:wptx1="http://www.garmin.com/xmlschemas/WaypointExtension/v1" xmlns:gpxtrx="http://www.garmin.com/xmlschemas/GpxExtensions/v3" xmlns:gpxtpx="http://www.garmin.com/xmlschemas/TrackPointExtension/v1" xmlns:gpxx="http://www.garmin.com/xmlschemas/GpxExtensions/v3" xmlns:trp="http://www.garmin.com/xmlschemas/TripExtensions/v1" xmlns:adv="http://www.garmin.com/xmlschemas/AdventuresExtensions/v1" xmlns:prs="http://www.garmin.com/xmlschemas/PressureExtension/v1" xmlns:tmd="http://www.garmin.com/xmlschemas/TripMetaDataExtensions/v1" xmlns:vptm="http://www.garmin.com/xmlschemas/ViaPointTransportationModeExtensions/v1" xmlns:ctx="http://www.garmin.com/xmlschemas/CreationTimeExtension/v1" xmlns:gpxacc="http://www.garmin.com/xmlschemas/AccelerationExtension/v1" xmlns:gpxpx="http://www.garmin.com/xmlschemas/PowerExtension/v1" xmlns:vidx1="http://www.garmin.com/xmlschemas/VideoExtension/v1">
  <time>2020-11-01T21:13:19.863Z</time>
  <bounds minlat="55.584890954" minlon="38.113379236" maxlat="55.598579356" maxlon="38.164458219"/>
  <trk>
    <name>2020-10-26 21:30:14</name>
    <trkseg>
      <trkpt lat="55.598527221" lon="38.122117287">
        <ele>139.020</ele>
        <time>2020-10-26T16:31:39Z</time>
        <course>0.000000</course>
        <speed>0.000000</speed>
      </trkpt>
      <trkpt lat="55.598579356" lon="38.122064313">
        <ele>139.130</ele>
        <time>2020-10-26T16:31:49Z</time>
        <course>330.141113</course>
        <speed>0.669203</speed>
      </trkpt>
      <trkpt lat="55.598536944" lon="38.122150563">
```

velocat, 2020-11-01:
> course and speed are only elements in gpx version 1.0, they were removed from gpx version 1.1. You can force gpx version 1.0, and then course and speed will be included if they exist:
That's what it was all about!  Thank you very much!

robertlipe, 2020-12-27:
https://github.com/gpsbabel/gpsbabel/pull/647 is marked. Closing.

closed 2020-12-27 by robertlipe