GPSBabel/gpsbabel #653: Linux Make libusb error

BenVella, 2020-12-02:
Libusb not building and does not appear to be in the directory given, despite the comments in code that we use our own LIBUSB.

Furthermore I installed the libusb-1.0.0-dev with the same result.

```
/Downloads/gpsbabel/gpsbabel-1.7.0/jeeps/gpslibusb.cc:35:14: fatal error: libusb-1.0/libusb.h: No such file or directory
   35 | #    include <libusb-1.0/libusb.h>
      |              ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/gpsbabel.dir/build.make:2598: CMakeFiles/gpsbabel.dir/jeeps/gpslibusb.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:104: CMakeFiles/gpsbabel.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

tsteven4, 2020-12-02:
Please see the INSTALL file, https://github.com/gpsbabel/gpsbabel/blob/master/INSTALL

It notes that cmake support is experimental, and recommends qmake.

qmake support is currently much more versatile with respect to the location of third party libraries.

cmake support is likely to improve and change significantly with Qt6.  However, Qt6 is currently has serious issues with the changes in qtextcodec support that may not be resolved for some time.

BenVella, 2020-12-03:
As mentioned, qmake works fine.

closed 2020-12-03 by BenVella