GPSBabel/gpsbabel #655: "Building from source" WEB page instructions not accurate for github clones.

bjkowalski, 2020-12-07:
The instructions on the "**Building from source**" WEB page (URL = http://www.gpsbabel.org/htmldoc-development/Source.html) state that building code is as simple as:

`./configure && make`

Those instructions work if you download the source tarball.  But if you clone the git repository (URL = https://github.com/gpsbabel/gpsbabel) there is no "configure" shell script to run.

Found the autoconf files in the **deprecated** directory but I didn't find any explanation of why they were moved there.  Perhaps I missed this discussion.

robertlipe, 2020-12-07:
It's now

qmake && make

I'll check the doc in htmldoc-development and update if needed. Thanks.

tsteven4, 2020-12-07:
Sorry.  The latest build instructions are in the file INSTALL.

bjkowalski, 2020-12-07:
On RedHat (maybe other Linux systems as well) I had to install qt5-qtbase-devel (see Issue #654).  With that yum install (and perhaps other Linux variants)  there is no qmake alias and you have to use qmake-qt5.

Running qmake-qt5 generated a Makefile which I could run.

So it appears on a RedHat Enterprise Linux 7 system the commands should be 

`qmake-qt5 && make`

robertlipe, 2020-12-14:
Distros are notorious for renaming things. Apparently qmake refers to a
version of Qt that hasn't been supported in > 15 years or so. They seem to
prefer abandoned and unmaintained projects over ones that are actively
developed. Redhat has a qtchooser that lets you change that and, of course,
using that makes you different from other RedHat users which is a wrech for
supporting that.

In short, we really can't document all the ways Linux distros will
randomize the naming and paths. Sorry. The situation is all pretty terrible
and we have to rely on Linux devs knowing more about the system they're
using than is commonly the case.

Also, to get ahead of this, Qt6.0 is out and it doesn't support features we
need. We have open bugreports open with Qt development and they'll be
restoring the functionality sometimes after 6.0.

So we have some work to do in this area for both code and doc, but Linux
development is just messy.