GPSBabel/gpsbabel #659: Consider 64bit executable for windows

tsteven4, 2020-12-28:
It is possible to fail memory allocation using the 32 bit windows executable with a large number of waypoints.

An example is "-i gpx -f big.gpx -o gpx -F xxx.gpx", where big.gpx contains 105,596 waypoints or thereabouts.
1. In this case std::bad_alloc is thrown and uncaught.  The program dies without an error message.  It would be easy to catch this exception and at least try to print a message.
2. With a 64 bit windows executable the correct output file is produced.

Using the MSVC 64 bit compiler generates many more warnings compared to the MSVC 32 bit compiler.  In general these have to do with the word size mismatch of the long type between the LLP64 memory model used by MSVC x86_amd64 and the LP64 memory model that was used to develop the majority of the code.  Note that CI builds with both MSVC 32 bit and 64 bit compilers, but historically we have always distributed only a 32 bit windows executable.

tsteven4, 2020-12-28:
Note that the QStarz BT-Q1000XT can log 400,000 points.

tsteven4, 2020-12-29:
Almost all the errors are C4267, The compiler detected a conversion from size_t to a smaller type.

Note that with  gcc and clang we do not enable Wconversion, Warn for implicit conversions that may alter a value.   If we enable this warning many of the warnings flagged by C4267 would be detected by gcc/clang.  Wconversion is not enabled by Wall.

I recommend we disable C4267 which brings the MSVC compiler warnings closer to gcc/clang.  The C4100 warning can easily be fixed as we have done in other places.

We currently appear to run Inno Setup in 32 bit mode and the resulting executables are placed in C:\Program Files (\x86), even for our 64 bit build.

MSVC 32 bit build, with C4267 disabled (2 warnings):
Severity	Code	Description	Project	File	Line	Suppression State
Warning	C4100	'hSHP': unreferenced formal parameter	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\shapelib\shpopen.c	3030	
Warning	C4477	'printf' : format string '%ld' requires an argument of type 'long', but variadic argument 2 has type 'time_t'	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\lowranceusr.cc	500	

MSVC 64 bit build, with  C4267 disabled (2 warnings):
Severity	Code	Description	Project	File	Line	Suppression State
Warning	C4100	'hSHP': unreferenced formal parameter	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\shapelib\shpopen.c	3030	
Warning	C4477	'printf' : format string '%ld' requires an argument of type 'long', but variadic argument 2 has type 'time_t'	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\lowranceusr.cc	500	

MSVC 32 bit build, with C4267 enabled (12 warnings):
Severity	Code	Description	Project	File	Line	Suppression State
Warning	C4100	'hSHP': unreferenced formal parameter	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\shapelib\shpopen.c	3030	
Warning	C4267	'initializing': conversion from 'size_t' to 'short', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\an1.cc	417	
Warning	C4267	'argument': conversion from 'size_t' to 'int16_t', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\an1.cc	425	
Warning	C4267	'argument': conversion from 'size_t' to 'int16_t', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\an1.cc	432	
Warning	C4267	'=': conversion from 'size_t' to 'short', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\an1.cc	446	
Warning	C4267	'=': conversion from 'size_t' to 'short', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\an1.cc	449	
Warning	C4267	'=': conversion from 'size_t' to 'short', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\an1.cc	456	
Warning	C4267	'=': conversion from 'size_t' to 'short', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\an1.cc	463	
Warning	C4267	'initializing': conversion from 'size_t' to 'short', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\an1.cc	527	
Warning	C4267	'initializing': conversion from 'size_t' to 'uint16_t', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\dg-100.cc	266	
Warning	C4477	'printf' : format string '%ld' requires an argument of type 'long', but variadic argument 2 has type 'time_t'	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\lowranceusr.cc	500	
Warning	C4267	'argument': conversion from 'size_t' to 'uint16_t', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\mmo.cc	1153	

MSVC 64 bit build, with C4267 enabled (119 warnings):
Severity	Code	Description	Project	File	Line	Suppression State
Warning	C4267	'function': conversion from 'size_t' to 'SAOffset', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\shapelib\dbfopen.c	867	
Warning	C4267	'=': conversion from 'size_t' to 'unsigned int', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\zlib\gzread.c	319	
Warning	C4267	'=': conversion from 'size_t' to 'unsigned int', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\zlib\gzread.c	400	
Warning	C4267	'=': conversion from 'size_t' to 'int', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\zlib\gzread.c	472	
Warning	C4267	'=': conversion from 'size_t' to 'unsigned int', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\zlib\gzwrite.c	212	
Warning	C4267	'=': conversion from 'size_t' to 'unsigned int', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\zlib\gzwrite.c	232	
Warning	C4267	'=': conversion from 'size_t' to 'int', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\zlib\gzwrite.c	371	
Warning	C4267	'function': conversion from 'size_t' to 'int', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\strptime.c	356	
Warning	C4267	'function': conversion from 'size_t' to 'int', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\strptime.c	357	
Warning	C4267	'function': conversion from 'size_t' to 'int', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\strptime.c	395	
Warning	C4267	'function': conversion from 'size_t' to 'int', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\strptime.c	396	
Warning	C4267	'function': conversion from 'size_t' to 'int', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\strptime.c	544	
Warning	C4267	'function': conversion from 'size_t' to 'int', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\strptime.c	545	
[... cut ...]
Warning	C4267	'return': conversion from 'size_t' to 'int', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\wbt-200.cc	466	
Warning	C4267	'argument': conversion from 'size_t' to 'int', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\wbt-200.cc	688	
Warning	C4267	'-=': conversion from 'size_t' to 'unsigned int', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\wbt-200.cc	957	
Warning	C4267	'+=': conversion from 'size_t' to 'unsigned int', possible loss of data	gpsbabel	C:\Users\strabert\source\repos\gpsbabel\wbt-200.cc	958

tsteven4, 2020-12-29:
Distributing only a 64 bit windows executable will preclude Windows XP and earlier versions, as well as x86 architectures.  Note that Qt 5.12 (and 5.9), which we have been using for releases only supports windows 7, 8.1 and 10 although it does support both  x86 and x86_64 architectures.  Thus moving to a 64 bit executable only drops support for x86 architecture and 32 bit versions of windows, but does not drop support for any 64 bit version of windows that we currently support.

robertlipe, 2020-12-30:
XP has been dead for a long time. I dont' care.

I agree. It looks like 4267 is creating a lot of low-value, hyperactive
noise.

We "fix" some of these warnings by deleting the code. I'm just tired of
looking at an1. Alan, Navigude and Netstumbler can be right behind it. I'd
have to think about mmo. Magellan might get some remaining life because
it's used internally by the magellanx code which may not be quite dead, but
it wouldn't take a lot of convincing to whack them both, too.

In all these cases where we've anguished about turning down a format, we've
not had any measurable uprising, so I think we're just torturing ourselves
with a lot of this effectively dead code.

What size is size_t if it's not an int? Is 'int' perhaps unsigned on this
combination and it's flipping out because size_t has to be signed to allow
for error values?

Footnote, on some systems time_t is a double. I dont' think that's true on
any system we care about. It's commonly a long long or a 64 bit, so 'int'
isn't always the best printf specificier. Solution: nuke more code using
time_t like I did earlier this week.

RJL

tsteven4, 2020-12-31:
In LLP64 and LP64 size_t is 64 bits, int is 32 bits.
https://en.wikipedia.org/wiki/64-bit_computing#64-bit_data_models
size_t is unsigned
https://en.cppreference.com/w/cpp/types/size_t

I agree, nuking dead formats is a good start.  If not streams may avoid the print specifier mismatch issue.  But since we running gcc/clang without Wconversion I just silenced C4267.  I suspect other tools are still complaining about the same issues.

tsteven4, 2020-12-31:
resolved by #660.  Note #660 does not catch std::bad_alloc, I will open another issue.

closed 2020-12-31 by tsteven4

tsteven4, 2021-01-27:
The OP has posted his use case:

`gpsbabel -t -i qstarz_bl-1000 -f G:/BIN Qstarz/Bus/191018_025410.BIN -f G:/BIN Qstarz/Bus/191018_030840.BIN -f G:/BIN Qstarz/Bus/191018_051311.BIN -f G:/BIN Qstarz/Bus/191018_060132.BIN -f G:/BIN Qstarz/Bus/191018_085648.BIN -f G:/BIN Qstarz/Bus/191018_103641.BIN -f G:/BIN Qstarz/Bus/191018_105524.BIN -f G:/BIN Qstarz/Bus/191022_062100.BIN -f G:/BIN Qstarz/Bus/191022_064623.BIN -f G:/BIN Qstarz/Bus/191022_070419.BIN -f G:/BIN Qstarz/Bus/191022_084050.BIN -f G:/BIN Qstarz/Bus/191022_105726.BIN -f G:/BIN Qstarz/Bus/191022_132113.BIN -f G:/BIN Qstarz/Bus/191022_140135.BIN -f G:/BIN Qstarz/Bus/191024_032619.BIN -f G:/BIN Qstarz/Bus/191024_034738.BIN -f G:/BIN Qstarz/Bus/191024_073459.BIN -f G:/BIN Qstarz/Bus/191024_115928.BIN -f G:/BIN Qstarz/Bus/191024_154910.BIN -f G:/BIN Qstarz/Bus/191025_035550.BIN -f G:/BIN Qstarz/Bus/191025_043545.BIN -f G:/BIN Qstarz/Bus/191025_071049.BIN -f G:/BIN Qstarz/Bus/191025_112810.BIN -f G:/BIN Qstarz/Bus/191122_141345.BIN -f G:/BIN Qstarz/Bus/191122_144235.BIN -f G:/BIN Qstarz/Bus/191122_155251.BIN -f G:/BIN Qstarz/Bus/191122_195534.BIN -f G:/BIN Qstarz/Bus/191122_231917.BIN -f G:/BIN Qstarz/Bus/191126_034830.BIN -f G:/BIN Qstarz/Bus/191126_040456.BIN -f G:/BIN Qstarz/Bus/191126_060736.BIN -f G:/BIN Qstarz/Bus/191126_065752.BIN -f G:/BIN Qstarz/Bus/191126_100211.BIN -f G:/BIN Qstarz/Bus/191126_113813.BIN -f G:/BIN Qstarz/Bus/191126_115852.BIN -f G:/BIN Qstarz/Bus/191223_045618.BIN -f G:/BIN Qstarz/Bus/191223_051448.BIN -f G:/BIN Qstarz/Bus/191223_084853.BIN -f G:/BIN Qstarz/Bus/191223_110433.BIN -f G:/BIN Qstarz/Bus/200119_095150.BIN -f G:/BIN Qstarz/Bus/200119_160610.BIN -f G:/BIN Qstarz/Bus/200124_145952.BIN -f G:/BIN Qstarz/Bus/200124_182019.BIN -f G:/BIN Qstarz/Bus/200124_202731.BIN -f G:/BIN Qstarz/Bus/200124_235952.BIN -f G:/BIN Qstarz/Bus/201008_034408.BIN -f G:/BIN Qstarz/Bus/201008_040959.BIN -f G:/BIN Qstarz/Bus/201008_053449.BIN -f G:/BIN Qstarz/Bus/201008_060914.BIN -f G:/BIN Qstarz/Bus/201008_063015.BIN -f G:/BIN Qstarz/Bus/201008_070605.BIN -f G:/BIN Qstarz/Bus/201008_081547.BIN -f G:/BIN Qstarz/Bus/201008_085349.BIN -f G:/BIN Qstarz/Bus/201008_090958.BIN -f G:/BIN Qstarz/Bus/201008_094117.BIN -f G:/BIN Qstarz/Bus/201008_101442.BIN -f G:/BIN Qstarz/Bus/201008_104257.BIN -f G:/BIN Qstarz/Bus/201008_111237.BIN -f G:/BIN Qstarz/Bus/201008_114608.BIN -f G:/BIN Qstarz/Bus/201008_120528.BIN -f G:/BIN Qstarz/Bus/201009_072230.BIN -f G:/BIN Qstarz/Bus/201009_073453.BIN -f G:/BIN Qstarz/Bus/201009_082512.BIN -f G:/BIN Qstarz/Bus/201009_092646.BIN -f G:/BIN Qstarz/Bus/201009_100100.BIN -f G:/BIN Qstarz/Bus/201009_102856.BIN -f G:/BIN Qstarz/Bus/201009_105850.BIN -f G:/BIN Qstarz/Bus/201009_114221.BIN -f G:/BIN Qstarz/Bus/201009_123105.BIN -f G:/BIN Qstarz/Bus/201009_125808.BIN -f G:/BIN Qstarz/Bus/201009_131443.BIN -f G:/BIN Qstarz/Bus/201009_132927.BIN -f G:/BIN Qstarz/Bus/201009_134804.BIN -f G:/BIN Qstarz/Bus/201009_140506.BIN -f G:/BIN Qstarz/Bus/201009_141727.BIN -f G:/BIN Qstarz/Bus/201009_142853.BIN -f G:/BIN Qstarz/Bus/201009_144432.BIN -f G:/BIN Qstarz/Bus/201009_150050.BIN -f G:/BIN Qstarz/Bus/201009_151259.BIN -f G:/BIN Qstarz/Bus/201009_153011.BIN -f G:/BIN Qstarz/Bus/201012_150135.BIN -f G:/BIN Qstarz/Bus/201012_151752.BIN -f G:/BIN Qstarz/Bus/201012_154338.BIN -f G:/BIN Qstarz/Bus/201012_161252.BIN -f G:/BIN Qstarz/Bus/201012_164038.BIN -f G:/BIN Qstarz/Bus/201012_171634.BIN -f G:/BIN Qstarz/Bus/201012_182724.BIN -f G:/BIN Qstarz/Bus/201012_185321.BIN -f G:/BIN Qstarz/Bus/201012_193550.BIN -f G:/BIN Qstarz/Bus/201012_202242.BIN -f G:/BIN Qstarz/Bus/201012_205355.BIN -f G:/BIN Qstarz/Bus/201012_215156.BIN -f G:/BIN Qstarz/Bus/201012_224741.BIN -f G:/BIN Qstarz/Bus/201016_125309.BIN -f G:/BIN Qstarz/Bus/201016_145855.BIN -f G:/BIN Qstarz/Bus/201016_160711.BIN -f G:/BIN Qstarz/Bus/201016_181614.BIN -f G:/BIN Qstarz/Bus/201016_192018.BIN -f G:/BIN Qstarz/Bus/201016_201900.BIN -f G:/BIN Qstarz/Bus/201030_045813.BIN -f G:/BIN Qstarz/Bus/201030_070149.BIN -f G:/BIN Qstarz/Bus/201030_075650.BIN -f G:/BIN Qstarz/Bus/201030_090319.BIN -f G:/BIN Qstarz/Bus/201030_105758.BIN -f G:/BIN Qstarz/Bus/201030_120040.BIN -f G:/BIN Qstarz/Bus/201030_125427.BIN -f G:/BIN Qstarz/Bus/201107_083327.BIN -f G:/BIN Qstarz/Bus/201107_084649.BIN -f G:/BIN Qstarz/Bus/201107_094647.BIN -f G:/BIN Qstarz/Bus/201107_110437.BIN -f G:/BIN Qstarz/Bus/201107_111634.BIN -f G:/BIN Qstarz/Bus/201107_124524.BIN -f G:/BIN Qstarz/Bus/201107_134028.BIN -f G:/BIN Qstarz/Bus/201107_140754.BIN -f G:/BIN Qstarz/Bus/201107_144150.BIN -f G:/BIN Qstarz/Bus/201107_154842.BIN -f G:/BIN Qstarz/Bus/201107_164147.BIN -f G:/BIN Qstarz/Bus/201108_141552.BIN -f G:/BIN Qstarz/Bus/201108_162853.BIN -f G:/BIN Qstarz/Bus/201108_180948.BIN -f G:/BIN Qstarz/Bus/201108_193514.BIN -f G:/BIN Qstarz/Bus/201108_200622.BIN -f G:/BIN Qstarz/Bus/201108_210056.BIN -f G:/BIN Qstarz/Bus/201108_212051.BIN -f G:/BIN Qstarz/Bus/201111_115615.BIN -f G:/BIN Qstarz/Bus/201111_121523.BIN -f G:/BIN Qstarz/Bus/201111_130840.BIN -f G:/BIN Qstarz/Bus/201111_141410.BIN -f G:/BIN Qstarz/Bus/201111_144921.BIN -f G:/BIN Qstarz/Bus/201111_161713.BIN -f G:/BIN Qstarz/Bus/201111_170241.BIN -f G:/BIN Qstarz/Bus/201111_181327.BIN -f G:/BIN Qstarz/Bus/201112_123708.BIN -f G:/BIN Qstarz/Bus/201112_125824.BIN -f G:/BIN Qstarz/Bus/201112_135852.BIN -f G:/BIN Qstarz/Bus/201112_151938.BIN -f G:/BIN Qstarz/Bus/201112_170015.BIN -f G:/BIN Qstarz/Bus/201112_184857.BIN -f G:/BIN Qstarz/Bus/201112_195047.BIN -f G:/BIN Qstar [... cut ...] IN Qstarz/Bus/201220_064755.BIN -f G:/BIN Qstarz/Bus/201220_082600.BIN -f G:/BIN Qstarz/Bus/201220_100553.BIN -f G:/BIN Qstarz/Bus/201220_114812.BIN -f G:/BIN Qstarz/Bus/201220_123226.BIN -f G:/BIN Qstarz/Bus/201220_133631.BIN -f G:/BIN Qstarz/Bus/201220_140921.BIN -f G:/BIN Qstarz/Bus/201220_151514.BIN -x nuketypes,waypoints -x track,title=%Y%m%d-%H%M -x position,distance=1F -x simplify,count=5000 -o kml,points=0,line_width=2,trackdata=0,labels=0,rotate_colors=1,prec=8 -F G:/BIN Qstarz/Bus.kml`

robertlipe, 2021-01-27:
I'll bet it's blowing up building the kml in core instead of streaming,.
But I fell for this once and spent hours analysing almist what the user
said and studied an irrelevant case.

Turn off points and lines. Turn on trackdata.

Provide full test cases...