GPSBabel/gpsbabel #690: enigma.test fails on i686

manfredu, 2021-02-12:
I'm trying to provide an up-to-date package of version 1.70 for Void Linux.

Besides some issues with cross-compiling I'm still trying to solve ("loader cannot find Qt library") a test from /testo.d/enigma.test fails on i686 only:

```
Binary files ./reference/enigma-gpsb.ert and /tmp/gpsbabel.5142/enigma.ert differ
ERROR comparing ./reference/enigma-gpsb.ert /tmp/gpsbabel.5142/enigma.ert
```

On x86_64 all tests run fine.

tsteven4, 2021-02-12:
Welcome.

The i686 enigma test failure was fixed in #690 and applied as a patch to Debian.

I need to do some research about the cross compilation issue, but I need to figure out what the best method was.

tsteven4, 2021-02-12:
As I mentioned in #621, your cross compilation issues will likely just go away with qmake instead of configure.  While homebrew doesn't do cross compilation they have switched their build to qmake, see https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/gpsbabel.rb.  It looks like #611 was the patch they needed to make this work.    Alternatively, you can try the debian cross patch https://sources.debian.org/patches/gpsbabel/1.7.0+ds-7/, however the debian cross patch will not work in the next release of gpsbabel as support for configure will have been removed.

tsteven4, 2021-02-12:
The documentation for the qmake options to use system libraries wasn't included in #611.  You can see it here
https://github.com/GPSBabel/gpsbabel/blob/master/INSTALL

manfredu, 2021-02-12:
> Welcome.

Thank you. :-)

> 
> The i686 enigma test failure was fixed in #690 and applied as a patch to Debian.

Found it, it's in #620.   

> 
> I need to do some research about the cross compilation issue, but I need to figure out what the best method was.

Thanks for the provided information.

manfredu, 2021-02-13:
Thanks for your help. Void Linux package gpsbabel 1.7.0 is available now.

closed 2021-02-13 by manfredu

tsteven4, 2021-02-13:
@manfredu thanks for making the latest gpsbabel release available on Void Linux.