GPSBabel/gpsbabel #694: track/route simplfy

alinagithub, 2021-02-14:
Thanks a lot for maintaining GPSBabel!
A suggestion, for the track/route simplify would be: force keep closest to waypoints. In addition, the closest track point to a waypoint could inheritate the name, desc... In other words, how do I simplify my track/route but preserve going through my waypoints. I did my own Python script for that, but would be great to see this in GPSBabel!
Thank you!
AL.

robertlipe, 2021-02-14:
Hi.

If you can code what you're describing (and describe it more clearly), you
can implement it as a filter on gpsbabel. Trackfilter.cc receives three
lists: waypoints, tracks, and routes. Modify those lists and you're done.
There are LOTS of examples that fine and other filter.cc files.

Send us a PR with test cases, docs, and code and we'll see about pulling it
in.

alinagithub, 2021-02-14:
Thank you Robert. I will definitely look at the filter.cc and give it a try.
Let me re-phrase the need, I hope more clearly: when exporting a KML from Google MyMaps, you get an out of order Waypoints, and a track with some quite high number of points - let a few thousands. What you can do today, is to reduce the density - deleting some point of track or route - but this do not preserve the key waypoints you absolutely want to pass by. What you want to provide to the GPS is probably a route made of the dozen of waypoints only, and let the GPS determine the best route between the waypoints, according to the traffic for example. Do do that, I build a route from waypoints re-order with the help of the track: I find the corresponding index of the closest point in the track. That gives me a compact route in a GPX that can be easily re-loaded in Kurviger, TomTom, etc...
Hope it clarifies, and think you for pointing me to the filter.cc!

robertlipe, 2021-02-14:
Thank you for that better explanation. We should probably unify our
parlance:

You get a track with a high number of trackpoints. Trackpoints are parts of
tracks and waypoints aren't. It sounds like you want to ensure that your
waypoints are part of the tracks (and I'd have to think about that if you
didn't actually pass over that exact spot on your previous trip) and use
those "knees" as control points in the track and ensure they don't get
optimized away as you then want to merge the remaining tracks and
waypoints, optimizing away most of the collected trackpointss, and then
turn that collected data into a GPX route.

We have a (super obscure) feature where you can set a route priority inside
a trackpoint and then the crosstrack error simplifier in the trackpoint
simplifier will treat those points as more special and try to not discard
them. "Sounds great! How do I use it?"  The sands of time have,
unfortunately, made that bit almost impossible to set.

There are dead formats that have set this bit for important control points
on read.  route_priority was set by the deprecated formats google, delbin,
and saroute. It's set by the arcdist filter in a way not apparent to me
right now. I *think* you could probably put a sledgehammer in the KML
reader that sensed if it was reading Placemarks (wpt_s) and just set the
route_priority member of that waypt or if it was on some attribute of the
KML (name starts with... or icon == or whatever) in those nearby wpt_foo
readers. Then use the existing filters to order ALL the points into a track
then maybe use another pass of the filter to reduce the track to the dozens
(?) of points that are still relevant.

It's certainly not a road well traveled (haha) in the code, but as as
one-off for a specific project where you control the data and don't mind
making a few manual tweaks and even a custom, hacked-up, GPSBabel, that's
all OK. That kind of thing is exactly why we're open source.

Good luck!
RJL

alinagithub, 2021-02-14:
Robert, this is exactly this. There is also another way I could do: a filter that keep trackpoint that are sufficiently close (radius) to a waypoints - thus waypoints act  as control points - but not exact spots. Thus you can make a route from a track using converting into routepoints these selected trackpoints only.
Thank a lot for your detailed answer!
B.R
AL.

robertlipe, 2021-02-14:
If you have millions or even thousands, this won't work, but if this is mostly a one-time, small-ish task, there's another form of brute force you can use:

Rename the waypoint in the order you want, but name them so they'll ascii-sort in the right order. ("1-Franklin Rd, "2, OHB@I-65N", "3 Exit93@Armory Dr", ...) Use our 'sort' filter to help sort by short name or other field.

Throw out the track data, possibly using our toss options.

Now that you've reduce the waypoints to some way you can sort them, use our transform filter, which will read waypoint (and everything else) in the order they appear and then convert to a route, which will make the routes with the route points in the order the waypoints were read.

We have a raft of data filters: http://www.gpsbabel.org/htmldoc-development/Data_Filters.html . I'd bet that if you're willing to tinker with your own data set and experiment with combining them (esp if you're willing to tear into the source a little and fiddle in ways like I described like setting the route_priority.

Good luck.

Meta. I think that there are two of us that read bugreuports on Github. We have several hundred subscribers to the mailing list, but the percentage that are actually active isn't great. Where should conversations like this happen? It started as "will you build me a thing" (no.) and morphed into "can we use the existing things to do this?" (maybe/probably).

alinagithub, 2021-02-15:
Robert, actually the renaming in MyMaps trick 0001-... + sort + discard is exactly what I end up to use with GPSBabel. Then indeed I can execute a second GPSBabel from this new file to generate a route with the ordered waypoints - for now I was simply loading waypoints only in Kurviger - then save back the route only. 
I downloaded the GPSBabel source code - I will try to setup and compile.
I definitely understand the (no) :)
Thanks again for your suggestions.
AL.

robertlipe, 2021-09-13:
Sounds like this reached closure. Glad we could help.

closed 2021-09-13 by robertlipe