GPSBabel/gpsbabel #708: DG-200: erase command not sent

hitausmomentti, 2021-04-11:
Downloading the data off a DG-200 works as expected with 1.7.0, but erasing it doesn't. According to the debug log the erase command is not sent to the device. With 1.6.0 the command is sent and the data is erased. The problem appeared both with the Ubuntu-packaged 1.7.0 as well as with the newest repo master compiled manually. I tried git bisect and it blamed the problem on commit a459fc2870da05a2b9d22a2befdc2435598d93bc. I tried various permutations of erase and erase_only with =1 and without.

Below logs from the newest version (69514646) and 1.6.0

```
sudo ./gpsbabel -D 2 -t -i dg-200,erase  -o gpx  /dev/ttyUSB0 /dev/null
GPSBabel Version: 1.7.0 
main: Compiled with Qt 5.14.2 for architecture x86_64-little_endian-lp64
main: Running with Qt 5.14.2 on Ubuntu 20.10, x86_64
main: QLocale::system() is en_GB
main: QLocale() is en_GB
main: QTextCodec::codecForLocale() is UTF-8, mib 106
options: module/option=value: dg-200/erase="1"
options: module/option=value: dg-200/erase_only="0" (=default)
options: module/option=value: gpx/snlen="32" (=default)
options: module/option=value: gpx/elevprec="3" (=default)
options: module/option=value: dg-200/erase="0" (=default)
options: module/option=value: dg-200/erase_only="0" (=default)
TX: Frame Start a0 a2 Payload_Len 0003 Cmd: getfileheader
RX: Start a0a2 Len 0005 Cmd: getfileheader
found 0 headers, nextheader=0
no further headers, aborting the loop
TX: Frame Start a0 a2 Payload_Len 0001 Cmd: getconfig
RX: Start a0a2 Len 002e Cmd: getconfig
options: module/option=value: gpx/snlen="32" (=default)
options: module/option=value: gpx/elevprec="3" (=default)
```

```
gpsbabel -D 2 -t -i dg-200,erase  -o gpx  /dev/ttyUSB0 /dev/null
GPSBabel Version: 1.6.0 
main: Compiled with Qt 5.12.5 for architecture x86_64-little_endian-lp64
main: Running with Qt 5.12.8 on Ubuntu 20.04.2 LTS, x86_64
main: QLocale::system() is C
main: QLocale() is C
main: QTextCodec::codecForLocale() is UTF-8, mib 106
options: module/option=value: dg-200/erase="1"
options: module/option=value: dg-200/erase_only="0" (=default)
options: module/option=value: gpx/snlen="32" (=default)
options: module/option=value: gpx/elevprec="3" (=default)
TX: Frame Start a0 a2 Payload_Len 0003 Cmd: getfileheader
RX: Start a0a2 Len 0005 Cmd: getfileheader
found 0 headers, nextheader=0
no further headers, aborting the loop
TX: Frame Start a0 a2 Payload_Len 0001 Cmd: getconfig
RX: Start a0a2 Len 002e Cmd: getconfig
TX: Frame Start a0 a2 Payload_Len 0003 Cmd: erase
RX: Start a0a2 Len 0005 Cmd: erase
cet_util: Converting from "US-ASCII" to "UTF-8", done.
```

tsteven4, 2021-04-11:
https://github.com/GPSBabel/gpsbabel/commit/a459fc2870da05a2b9d22a2befdc2435598d93bc would appear to be unrelated.

Can you please attach the results of (debug level changed)
gpsbabel -D 5 -t -i dg-200,erase  -o gpx  /dev/ttyUSB0 /dev/null
with https://github.com/GPSBabel/gpsbabel/commit/695146460d35a5e00efde5b6c2c2cf767181294c

hitausmomentti, 2021-04-11:
Here you go, 
```
((69514646...))$ sudo ./gpsbabel -D 5 -t -i dg-200,erase  -o gpx  /dev/ttyUSB0 /dev/null 
options: module/option=value: dg-200/erase="1"
options: module/option=value: dg-200/erase_only="0" (=default)
options: module/option=value: gpx/snlen="32" (=default)
options: module/option=value: gpx/elevprec="3" (=default)
options: module/option=value: dg-200/erase="0" (=default)
options: module/option=value: dg-200/erase_only="0" (=default)
gbser_init("/dev/ttyUSB0")
options: module/option=value: gpx/snlen="32" (=default)
options: module/option=value: gpx/elevprec="3" (=default)
```

And 1.6.0 for comparison.  
```
GPSBabel Version: 1.6.0 
main: Compiled with Qt 5.12.5 for architecture x86_64-little_endian-lp64
main: Running with Qt 5.12.8 on Ubuntu 20.04.2 LTS, x86_64
main: QLocale::system() is C
main: QLocale() is C
main: QTextCodec::codecForLocale() is UTF-8, mib 106
Sent: a0 a2 00 03 bb 00 00 00 bb b0 b3 
TX: Frame Start a0 a2 Payload_Len 0003 Cmd: getfileheader
Receiving a0 a2 00 05 bb 00 00 00 00 00 bb b0 b3 RX: Start a0a2 Len 0005 Cmd: getfileheader

found 0 headers, nextheader=0
no further headers, aborting the loop
Sent: a0 a2 00 01 b7 00 b7 b0 b3 
TX: Frame Start a0 a2 Payload_Len 0001 Cmd: getconfig
Receiving a0 a2 00 2e b7 01 00 00 00 00 00 00 00 00 00 00 00 00 75 30 00 00 27 10 00 00 13 88 f0 b8 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 00 00 01 01 03 dd b0 b3 RX: Start a0a2 Len 002e Cmd: getconfig

Sent: a0 a2 00 03 ba ff ff 02 b8 b0 b3 
TX: Frame Start a0 a2 Payload_Len 0003 Cmd: erase
Receiving a0 a2 00 05 ba 00 00 00 01 00 bb b0 b3 RX: Start a0a2 Len 0005 Cmd: erase
```

tsteven4, 2021-04-11:
thanks.  bisect was correct.  a workaround is to not use positional parameters, e.g.
./gpsbabel -D 5 -t -i dg-200,erase **-f /dev/ttyUSB0** -o gpx  **-F /dev/null**

hitausmomentti, 2021-04-11:
Thank you! The workaround helped.

PR #709 (2021-04-13): restore lost format options when using positional parameters
This resolves #708
closed 2021-04-13 by tsteven4