GPSBabel/gpsbabel #712: Feature request: support RINEX file

Jamesits, 2021-05-04:
I'm not sure if it is technically possible (I'm not familiar with GIS formats), and I'm not sure why there are no issues or pull requests referencing this format, but is it possible to support RINEX files (at least as a source format)?

My use case: My Garmin GPSMAP 66s can record gpx/fit format but the elevation data recorded is derived from the barometer rather than the GPS. In some cases (e.g. inside the pressurized cabin of a plane) this elevation data is basically garbage. This unit have the option to record a separate RINEX file, but none of the official software supports reading it. Based on its format I guess it is possible to recover the correct elevation data (again, I'm not familiar with GIS formats at all so correct me if I'm wrong). If everything's correct, then it will be more accurate to convert the RINEX files to GPX rather than relying on the unit's own GPX files.

robertlipe, 2021-05-04:
RINEX contains a lot of data that isn't really pertinent in our world. When
GIS pros think of "converting"  RINEX, they're doing numeric analysis on
the data beyond what we can (or will) do. The infrequency of the requests
on the mailing list tells us there isn't a pent up demand for this.

Garmin has struggled with this for years. Fifteen years ago, they added a
setting to help.
https://forums.geocaching.com/GC/index.php?/topic/129478-60csx-elevation/.
In a close sibling to your model, it looks like there's a setting for auto
calibrate from GPS alt that makes the baro snap to the GPS setting every N
minutes.
https://forums.garmin.com/outdoor-recreation/inreach/f/gpsmap-66i/215140/wrong-elevation-position-not-always-recorded-in-auto,
but for years and years, they've staunchly refused to add a setting to just
disable the baro.
https://outdoors.stackexchange.com/questions/22893/systematic-and-random-error-of-gps-or-barometric-altimeter-how-to-optimally-use

For what you'd pay a contract programmer to build a RINEX reader, you can
buy an inexpensive data logger. You may even be able to find a geocacher
that'll trade your 66s with a 66 that simply doesn't have it.

Sorry that these are probably not the answers you're looking for.

RJL

Jamesits, 2021-05-05:
Spent half a day reading RINEX specs and now I understand why nobody want to implement a RINEX reader. That format literally found the worst point between the ability to be parsed by a program and the ability to be read by a human, plus it requires external data and a lot of math (well, more than I'm capable to do) to actually get the coordinates. 

The "auto calibrate" settings didn't work for my specific use case. I had it turned on, but I could see it easily got confused. It is really designed only to compensate air pressure changes due to (not too violent) weather changes. I don't understand why Garmin doesn't just let me turn off the baro, since my unit can display the GPS elevation on the screen, it is just not possible to record it into the GPX files. I guess next time I'll just harvest some NMEA data out of this unit's USB port and hope it can give me the correct readings. 

Really thank you for the info.

closed 2021-05-05 by Jamesits