GPSBabel/gpsbabel #715: Cannot compile without USB support: "libusb.h: No such file or directory"

richlv, 2021-05-28:
gpsbabel 1.7.0.
OpenSUSE Tumbleweed.

```
./configure --without-libusb
qmake-qt5
make
...
g++ -c -pipe -O2 -std=gnu++1y -Wall -Wextra -D_REENTRANT -fPIC -DNDEBUG -DHAVE_UNISTD_H -DHAVE_STDARG_H -DHAVE_LIBUSB_1_0 -DMAXIMAL_ENABLED -DFILTERS_ENABLED -DSHAPELIB_ENABLED -DCSVFMTS_ENABLED -DQT_NO_DEBUG -DQT_CORE_LIB -I. -Izlib -Ijeeps -I/usr/include/qt5 -I/usr/include/qt5/QtCore -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o gpslibusb.o jeeps/gpslibusb.cc
jeeps/gpslibusb.cc:35:14: fatal error: libusb-1.0/libusb.h: No such file or directory
   35 | #    include <libusb-1.0/libusb.h>
      |              ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:4184: gpslibusb.o] Error 1
```

robertlipe, 2021-05-28:
We've removed configure support after 1.7.0, so there will be no change here.

I'd advise against fighting this. Just install libusb-devel (or whatever it's called on your OS) and move on. The tens of kilobytes you might save just aren't worth being off the beaten path.

closed 2021-05-28 by robertlipe

tsteven4, 2021-05-28:
If your packaging 1.7.0 for open suse we can help you patch using commits from after 1.7.0.  The patches will require you to use qmake instead of configure, as will our next release.

If your just trying to compile for your own use then @robertlipe suggestion is easier.

richlv, 2021-05-29:
Thank you for the quick reply. Will install the dependency then.
I'm not the packager, but I have asked the packagers whether they will have a chance to update from 1.5.4.