GPSBabel/gpsbabel #717: Golang language bindings

SchoolGuy, 2021-07-31:
I am starting to develop an application in Go and was thinking if it is possible if you could create language bindings for go? Calling an executable is always nasty since it requires the consumer to parse the output and this is error prone.

To give a hint how you could start and what I am think of I will link you a blog entry of the Go Blog where there is a small introduction tutorial done for this: https://blog.golang.org/cgo

robertlipe, 2021-08-03:
Hi..As described in the FAQ, we're not interested in maintaining a library.

Parsing is fast and error-free with our  gpx (xml) and GeoJSON formats. We
support pipes and compression. Our xcsv and custom style formats make it
easy to pick just the fields you care about.

Sorry if that's not the answer you're looking for, but here we are.

RJL

SchoolGuy, 2021-08-03:
Okay I did not scroll down that far... Sorry to bother you. Thanks for the honest answer and thanks for keeping this project alive!

closed 2021-08-03 by SchoolGuy