GPSBabel/gpsbabel #719: Locations.fit on Garmin devices

fabzzap, 2021-09-11:
Garmin Edge Explore devices, and possibly more Garmin Edge device models, store saved locations in a file called Locations.fit .

I wanted to convert such a file to a GPX with only waypoints (section `wpt`), but GPSBabel produced an empty GPX file.
```<?xml version="1.0" encoding="UTF-8"?>
<gpx version="1.0" creator="GPSBabel - https://www.gpsbabel.org" xmlns="http://www.topografix.com/GPX/1/0">
  <time>2021-09-11T14:10:32.576Z</time>
  <trk/>
</gpx>
```
Digging into the code, it turns out that the `global_id` of that file is 0x1d (29), and garmin_fit.cc's code does not support it.

Garmin makes a FIT SDK available for download on a restrictive licence. While I couldn't find anything in that licence that prohibits working on open-source software to support the FIT format, I'd like to know others' opinion about that before downloading it (which requires accepting the licence).

Also, more in general, are there other ways to approach the developers? I couldn't find an IRC channel, and, while the Sourceforge page looks abandoned, the mailing list gpsbabel-misc@sourceforge.net had posts as late as June 2021, which makes its active/abandoned status unclear.

tsteven4, 2021-09-11:
Thanks for checking about the license.  We have rejected the fit sdk license before, e.g. see https://sourceforge.net/p/gpsbabel/mailman/message/36145201/  If you want to discuss this further @robertlipe needs to make the call.

sourceforge is abandoned except for the gpsbabel-misc, gpsbabel-code, gpsbabel-announce.  See https://www.gpsbabel.org/lists.html .  Other than that communication is via github as you found.

GPSBabelDeveloper, 2021-09-11:
Hi, and thanx for asking.

Restricted licenses just don't work for us. It's not like we can enforce
people to accept a third party license when they download a copy that we
include.

As for the quietness on our lists, I've noticed that. Our download and use
counters have dropped from peak, but this tracks use of dedicated gps in
general. We're a twenty year old program and have reached a quite stable
base of code and users.

fabzzap, 2021-09-12:
For sure one can't copy any code from the FIT SDK into an open-source project. But: are those who accepted the FIT SDK licence banned from developing FIT support in GPSBabel, even if they just reimplement the protocol with 100% new code?

If the answer is "yes", I will forget about accepting it (which I haven't done yet) and try and reverse-engineer that specific file.

robertlipe, 2021-09-13:
It's been a long time since I looked at the license, but there was something about it that was so restrictive, that I refused to accept it. 

Inviting legal problems from a company as well armed as Garmin is worth avoiding.

Our code in garmin_fit.cc had a pretty rough landing, but over time it's seemingly become robust enough for most use. It's likely that it contains some hints (Math_Semi_To_Deg(), etc.) that may apply to that file for a clean room study.

fabzzap, 2021-11-13:
I mostly reverse-engineered the format and opened pull request #762 .

I haven't downloaded the FIT SDK, eventually it was not necessary

tsteven4, 2021-11-19:
This was fixed by #762

closed 2021-11-19 by tsteven4