GPSBabel/gpsbabel #720: Duplicating name tag data

macias, 2021-09-12:
I noticed it on several occasions, but the simplest case is splitting .gpx file which contains elements (for example waypoints) with "name" tag. After splitting file those elements not only will have "name" tag, but also "cmt" and "desc" filled with the info from "name".

It is problematic because it means splitting one file into two halves produces files bigger than original.

robertlipe, 2021-09-13:
This is Working as intended. GPX readers may read names from any of those three fields, so if we have nothing to put in them, we duplicate the field we do have.

If you'd like to make GPSBabel not do that, find waypt_add() in waypt.cc and look at the end for the two blocks for shortname.isNull() and description.isEmpty() and just comment them out. That should neutralize this.

closed 2021-09-13 by robertlipe

macias, 2021-09-14:
Thank you for the reply and clarification, but it is odd choice -- after all user is splitting file to get some sample and the program i splitting it AND altering it. If the original file was good for the user, why to "improve" it? Splitting and filling the tags are separate concerns and should be treated as such.