GPSBabel/gpsbabel #754: xmapwpt.xml document lost

tsteven4, 2021-11-03:
It was deleted here https://github.com/GPSBabel/gpsbabel/commit/25473724b087ccdc077289eba79904a9c9ed7545 in an apparently unrelated changeset, but the format is still active.  It is a style file, so it is low maintenance.

Do we want to delete the style or restore the lost document?

robertlipe, 2021-11-05:
I could go either way, but this was pretty clearly a screwup of mine. Sorry.

PocketPC is clearly long dead - and was barely alive when it wasn't - but this wasn't a conscious decision to kill it. Maybe we should escalate that decision since I apparently wounded it.

If agreed, do you want to drop the .style file or shall I?

tsteven4, 2021-11-05:
I'll kill it.

tsteven4, 2021-11-05:
Shall we kill xmap and xmap2006 as well?

robertlipe, 2021-11-05:
Yes, please. Delorme mapping products are long dead.

tsteven4, 2022-01-17:
closed via #757

closed 2022-01-17 by tsteven4