GPSBabel/gpsbabel #827: GPI file with bitmaps doesn't display properly.

dpawlyk, 2022-01-22:
GPI file with bitmaps doesn't display properly. They show up as white-ish with a dim outline of item.

Not sure if it's my bitmaps or something else. I was told by the author of GPXSee it was an issue with the palette.

The attached files include the inputs and the command to generate the output.

[BTCNJ-Locations.zip](https://github.com/GPSBabel/gpsbabel/files/7919367/BTCNJ-Locations.zip)

tumic0, 2022-01-23:
The original issue is here: tumic0/GPXSee#423

tsteven4, 2022-01-23:
If you give a bitmap we use it for all points when we write the gpi file.  Not that this is what you want, but I did
 ./gpsbabel -i gpx -f "Starts.gpx" -o garmin_gpi,bitmap="poiblockgreen.bmp" -F "starts.gpi"
 ./gpsbabel -i gpx -f "Stops.gpx" -o garmin_gpi,bitmap="poiblockred.bmp" -F "stops.gpi"
with some modified code trying to address the color problem.  @dpawlyk Can you check and see if starts.gpi and stops.gpi display properly?
[gpi.zip](https://github.com/GPSBabel/gpsbabel/files/7921025/gpi.zip)

@tumic0 can you tell me why you always set the high order byte to 0xff in bgr2rgb and rgb(quint32 r, quint32 g, quint32 b)?  I see some old documents that show this extra byte as 0x00, and it appears our code assumed that.

tumic0, 2022-01-23:
> can you tell me why you always set the high order byte to 0xff in bgr2rgb and rgb(quint32 r, quint32 g, quint32 b)? I see some old documents that show this extra byte as 0x00, and it appears our code assumed that.

Because the formats (where this function is used) do not support the alpha channel, but Qt does in QRgb. A color with the alpha channel set to 0x00 is full transparent regardless of what R,G and B is set to. To get an opaque RGB color in Qt, you have to use 0xFF as the alpha channel.

tumic0, 2022-01-23:
Btw., the files from gpi.zip are still corrupted, the palette is still not properly written to the file.

tsteven4, 2022-01-23:
> Btw., the files from gpi.zip are still corrupted, the palette is still not properly written to the file.
Thanks.  Here they are with the extra byte in the palette as zeros.
Unfortunately,  I don't have a way to look at these myself.

[gpialpha0.zip](https://github.com/GPSBabel/gpsbabel/files/7921132/gpialpha0.zip)

dpawlyk, 2022-01-23:
Those GPI files basically work. The magenta background isn't being treated as transparent though. I wasn't sure if the device would read more than one GPI file (it does).
 
[image]

Might be useful if one could change the bitmap midstream.

tsteven4, 2022-01-23:
did you try gpi.zip or gpialpha0.zip?

dpawlyk, 2022-01-23:
Both look the same.

tumic0, 2022-01-23:
Still corrupted. You need to write the whole 256 item palette, if you set that palette size in the image header. In your files, only ~40 palette items are written.

tsteven4, 2022-01-23:
@tumic0 do you have a documentation source for the gpi format that told you how to decode the image information (id, height, width, linesize, bpp, ...) in gpiparser.cc readSymbol?  We don't seem to agree.  I can't defend our interpretation, I am not familiar with the development of our gpi support.

tumic0, 2022-01-23:
Well, the primary source, as always in case of Garmin, is reverse engineering. So if you have  a Garmin unit, you have the same "documentation" source as I have...

But apart from that "ultimate" format specification, there are many Garmin produced GPI files out there which all have it right and even 3rd party SW that produces GPI files that also write correct GPI files. In some cases even GPSBabel writes valid GPI files (depending on the image) but not always...

tumic0, 2022-01-23:
Now it works. Only the transparency is broken, but that is because of the images, that have the "transparent" color set to 0xDC00FF instead of 0xFF00FF.

dpawlyk, 2022-01-23:
Is that from my bitmaps?

tumic0, 2022-01-23:
> Is that from my bitmaps?
yes, look at them in some image editor like Gimp and you will see, that the magenta borders are in fact not "real" magenta.

tsteven4, 2022-01-23:
BTW this in process fix missed our 1.8.0 release by four days.  However, we still don't support using different bitmaps for different points.

dpawlyk, 2022-01-23:
Not a real problem that it missed the last release. That it will get out sometime is good enough.

The multiple bitmaps is a "might be nice" feature.

Thanks.

tumic0, 2022-01-23:
Even with the old and broken GPSBabel versions you can produce valid GPI files with symbols - you just have to use images with a full 256 color palette.

tsteven4, 2022-01-23:
yes.  thanks for your help @tumic0.

dpawlyk, 2022-01-24:
Alas, on Garmins that I've tried (1030+, 1000), the two files don't work. Two categories show up but the names are repeated and the POIs from the second file show up as blank.

tumic0, 2022-01-24:
Well, I was a little bit afraid this can happen. This is why I would rather fix the issue by always writing the full 256 items palette, than to write it only partially and set the palette size. While it is clear, that the original implementation is broken - if you do not write the palette size and write only a variable size palette, how can the reader know, how many items to read? - it is not clear if ALL Garmin SW can handle partial palettes (we only know, that some of their SW can do that). And evidently it does not.

There may of course be another problem with the files, I have no idea what all has been changed when producing those two GPI files, but the patch does not seem to alter much more...

dpawlyk, 2022-01-24:
I wasn't clear enough. There isn't any problem with the files. Each of them works properly if one is used at a time.

The Garmin software doesn't work properly if there are more than one file. The category of the first file is repeated and the POIs in the second file are listed but are blank. 

The Garmin POI loader software can take multiple input files and create one gpi file (each of the input file names becomes the category of the POIs in it). That works fine.

tumic0, 2022-01-24:
Ok, that's something completely different, if the files do not work together (but work separate). That will be caused by the clash of the category/symbol ids.

dpawlyk, 2022-01-24:
I think it won't be a problem with the future version of GPS Babel.

tsteven4, 2022-01-25:
Does using the category option allow the files to work together?

./gpsbabel -i gpx -f Starts.gpx -o garmin_gpi,bitmap=poiblockgreen.bmp,category=Starts -F Starts.gpi
./gpsbabel -i gpx -f Stops.gpx -o garmin_gpi,bitmap=poiblockred.bmp,category=Stops -F Stops.gpi

dpawlyk, 2022-01-25:
I don't think there's a problem _unless_ you are using a bitmap (like the ones I'm using) with the current version of GPSBabel. The files you provided (using the updated version of GPSBabel) above work properly.

On a different/unrelated topic...

POI Loader lets you create a single file (with multiple bitmaps and categories).

tsteven4, 2022-01-25:
Here are the files when I specify the categories "Starts" and "Stops" as I described above.
[try4.zip](https://github.com/GPSBabel/gpsbabel/files/7933844/try4.zip)
Does this resolve

> Alas, on Garmins that I've tried (1030+, 1000), the two files don't work. Two categories show up but the names are repeated and the POIs from the second file show up as blank.

dpawlyk, 2022-01-25:
Yes, those files work.

PR #828 (2022-01-26): correct color palette issues in gpi writer.
This resolves #827
closed 2022-01-26 by tsteven4