GPSBabel/gpsbabel #832: tests fail on sparc64

tsteven4, 2022-01-24:
@jspricke  shows test failures on sparc64: https://buildd.debian.org/status/package.php?p=gpsbabel&suite=sid

These all seem to be problems with the unicsv reader reading Time or utc_t.  In all cases we get seem to only get the date.

This would be much easier if we somehow had access to a sparc64 machine.  Does anyone know of any publicly available resources?

jspricke, 2022-01-24:
***@***.***  shows test failures on sparc64: https://buildd.debian.org/status/package.php?p=gpsbabel&suite=sid
>
>These all seem to be problems with the unicsv reader reading Time or utc_t.  In all cases we get seem to only get the date.
>
>This would be much easier if we somehow had access to a sparc64 machine.  Does anyone know of any publicly available resources?

You could ask at Debian:

https://db.debian.org/machines.cgi?host=kyoto

(currently it seems to be down, otherwise I could run tests for your.)

In general: This does not really look like a problem in gpsbabel as it 
works fine on all other architectures in Debian. Also sparc64 is not a 
release architecture. So don't feel inclined to debug this.

tsteven4, 2022-01-25:
@jspricke I got gpsbabel-gui_1.8.0+ds-3_amd64.deb  gpsbabel_1.8.0+ds-3_amd64.deb installed in a docker container.

languages look good😀.  I tested french and I saw the Qt translations and both sets of gpsbabel translations in use.

gmapbase looks good😀.  I used the mappreview feature and it worked.  however, because you are embedding it in the application you don't really need your patch gmapbase.html-location, /usr/share/gpsbabel/gmapbase.html doesn't exist.  The patch does no harm, it isn't found so we look in the Qt resource system and find it.

the help in the gui is only partially functional🙁  This is at least partly due to  your patch htmldoc-location.  The main help page works, we find /usr/share/doc/gpsbabel/gpsbabel.html.  

But all the other help does not work.  For example, hitting the help button on an input option dialog will generate a url like file:///usr/share/doc/gpsbabel/fmt_gpx.html, which does not exist.  Hitting help on the filter option dialog generates a url like file:///usr/share/doc/gpsbabel/Data_Filters.html, which does not exist.  I don't think our gpsbabel.org target which we use to generate the code for www.gpsbabel.org is going to help directly.  www.gpsbabel.org does a bunch of template voodoo that will probably hang you up.

Hitting a help button when firefox is already running generates a popup "Firefox is already running, but is not responding. To use Firefox, you must first close the existing Firefox process, restart your device, or use a different profile."  I don't see this error when I build our unpatched code on Ubuntu focal, I am not sure what is up with that.  On my build the second help request opens a new tab in firefox.  I can duplicate the "Firefox is already running ..." issue with multiple firefox requests outside gpsbabel.  In my container
firefox /tmp&
firefox /usr/share&
generates the popup.  In my host is does not.

I do see messages with your version:

> QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-tsteven4'
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-tsteven4'
Gtk-Message: 16:06:55.458: Failed to load module "canberra-gtk-module"
Gtk-Message: 16:06:55.460: Failed to load module "canberra-gtk-module"
[GFX1-]: glxtest: libpci missing

tsteven4, 2022-01-25:
removing the three {...} lines from xmldoc/babelmain.xsl and commenting out the last three lines of tools/make_gpsbabel_org.sh, then "make gpsbabel.org" will get you close, but you gone outside what we support.

```
-  {extends file="main.tpl"}
-  {block name=body}{literal}
-      {/literal}{/block}
```
```
+#tools/fixdoc "${web}/htmldoc-${docversion}" "GPSBabel ${docversion}:"
+#tools/mkcapabilities "${web}" "${web}/htmldoc-${docversion}"
+#cp gpsbabel.pdf "${web}/htmldoc-${docversion}/gpsbabel-${docversion}.pdf"
```

jspricke, 2022-01-25:
***@***.*** I got gpsbabel-gui_1.8.0+ds-3_amd64.deb  gpsbabel_1.8.0+ds-3_amd64.deb installed in a docker container.
>
>languages look good😀.  I tested french and I saw the Qt translations and both sets of gpsbabel translations in use.

Works for me as well now (had LC_MESSAGES set before).

> gmapbase looks good😀.  I used the mappreview feature and it worked.  however, because you are embedding it in the application you don't really need your patch gmapbase.html-location, [...]

Dropped, thx.

> Hitting a help button when firefox is already running generates a popup "Firefox is already running, but is not responding. To use Firefox, you must first close the existing Firefox process, restart [...]

I'm not able to reproduce that, maybe because you use Docker?

>removing the three {...} lines from xmldoc/babelmain.xsl and commenting out the last three lines of tools/make_gpsbabel_org.sh, then "make gpsbabel.org" will get you close, but you gone outside what we support.

Thanks a lot for this! Done in -4.