GPSBabel/gpsbabel #876: When converting to fit format, a name should be generated if there's none in the original format

ari-s, 2022-05-01:
It looks like fit files have a name inside them. At least some devices delete `fit`-files if that is not set.

Therefore, in the case the original file does not have a corresponding value I suggest to generate one, e.g. from the filename (maybe discarding the extension)

gpx files can have a `name` tag directly below the `gpx` root. <br/>
When I convert such a file to `fit` and put it on my garmin watch, the watch deletes the file. This should be avoided, hence the suggested workaround.

tsteven4, 2022-12-08:
Writers can see session information.  In the common use case of one input file I think we should be able to recover the input file name.

tsteven4, 2022-12-08:
It isn't clear to that the name in the course message is what the OP needs on his watch.