GPSBabel/gpsbabel #880: TimeSpan don't import from KML file

maximecharriere, 2022-05-24:
> ### Reopening of #482 

## Issue

When you export a [Google Maps Timeline](https://www.google.com/maps/timeline) to KML with the `Export this day to KML` button (clic on the gear at the bottom right), you get à KML file with waypoints and track containing time values. However GPSbabel fails to import time infos because the KML file is invalide (see well documented issue #482) . 
You can see in the [XSD file](http://schemas.opengis.net/kml/2.2.0/ogckml22.xsd) that the `AbstractTimePrimitiveGroup` element (Timespan, TimeStamp, etc.) must appear before the `AbstractGeometryGroup` element (Point, LineString, etc.) in the kml file and Google did the invers (tks Google...).  
It would be great if Google made an update, but I don't really believe it would append.  

I will use kml from Google Maps Timeline a lot in the future. So I would like to add to GPSbabel the understanding of kml with inverted LineString and Timespan elements.  

## Question

I just downloaded the gpsbabel source code and read the `kml.h` and `kml.cc` files. However, I don't understand where the limitation of the tag order comes from.  
**Can anyone guid me on which part of the code can cause the limitation of the tag order?**

### Releatd issues/PRs found: #115 #482 #484 #516 #518

robertlipe, 2022-05-24:
The limitation is that we don't support broken KML. Even if I could point
you to a few changes that might work, there may be other changes needed as
it's not gone through our QA. Even if you presented us with such a change,
ew wouldn't accept it as we're a format converter and not a format fixer.

Sorry. The onus is on getting the Big Guys to fix their defective code.

RJL

maximecharriere, 2022-05-24:
Okay, I understand.  
I will code a small script to correct the kml file and post it here if future users have the same problem.

maximecharriere, 2022-05-24:
Small Python3 script to fix KML files generated from  [Google Maps Timeline](https://www.google.com/maps/timeline)
```Python
import xml.etree.ElementTree as ET

def fixKml(inputFile, outputFile):
    namespace = {'':"http://www.opengis.net/kml/2.2"}
    ET.register_namespace('',"http://www.opengis.net/kml/2.2")
    ET.register_namespace('gx', "http://www.google.com/kml/ext/2.2")

    tree = ET.parse(inputFile)
    root = tree.getroot()

    for placemark in root[0].findall('Placemark',namespace):
        geoElem = placemark.find('LineString',namespace)
        if not geoElem:
            geoElem = placemark.find('Point',namespace)
        placemark.append(geoElem)
        placemark.remove(geoElem)

    tree.write(outputFile, encoding='UTF-8',method = 'xml',xml_declaration=True)

if __name__ == "__main__":
    fixKml('history-2022-04-15.kml', 'history-2022-04-15_fixed.kml')
```

closed 2022-05-24 by maximecharriere