GPSBabel/gpsbabel #887: simplify fails on track

simsala, 2022-06-25:
I am trying to simplify tracks.
This used to work but now fails:
        gpsbabel -t -i gpx -f my.gpx -x track,merge,discard,pack,simplify,count=1000 -o gpx -F new.gpx
This gives me an error message: "'simplify' is an unknown option to track."
Event though simplify is described to work on tracks:
https://www.gpsbabel.org/htmldoc-1.8.0/filter_simplify.html

Any specific reason for this?
Thanks

tsteven4, 2022-06-25:
Your syntax is incorrect.  This is not a change.  Your are using two filters, track and simplify.  Each needs to be invoked with it's own -x.
gpsbabel -t -i gpx -f my.gpx -x track,merge,discard,pack -x simplify,count=1000 -o gpx -F new.gpx

simsala, 2022-06-25:
Thanks. I was using an old script that used to work...
Nevertheless using your syntax fixes the issue.
Thanks a lot.

closed 2022-06-25 by simsala

tsteven4, 2022-06-25:
The behavior changed in https://github.com/GPSBabel/gpsbabel/commit/259afe8ad, which was not in 1.6.0, but was in 1.7.0 and 1.8.0. 
```
Fix validation of options supplied with a format or filter.
Previously unknown options were only detected if all given options
were unknown.  Now any unknown option will be reported, irregardless
of any other options supplied.
The behavior of ignoring unknown options is unchanged, they are ignored
except for the warning.
```
So in 1.6.0 you didn't see a warning, but the simplify filter was not run!
You can verify this by using the validate filter, with your command line note the track waypt count is not 10.
In 1.6.0:
```
:~/work/gpsbabel-gpsbabel_1_6_0.1$ ./gpsbabel -t -i gpx -f reference/track/trackfilter.gpx -x track,merge,discard,pack,simplify,count=10 -x validate,debug -o gpx -F new.gpx

Processing waypts
point ct: 0, waypt_count: 0

Processing routes
route head ct: 0, route_count: 0
total route point ct: 0, route_waypt_count: 0

Processing tracks
track 0 ct: 64, waypt_count: 64
track head ct: 1, track_count: 1
total track point ct: 64, track_waypt_count: 64
```
But with the correct syntax to invoke the filter the count is 10:
```
:~/work/gpsbabel-gpsbabel_1_6_0.1$ ./gpsbabel -t -i gpx -f reference/track/trackfilter.gpx -x track,merge,discard,pack -x simplify,count=10 -x validate,debug -o gpx -F new.gpx

Processing waypts
point ct: 0, waypt_count: 0

Processing routes
route head ct: 0, route_count: 0
total route point ct: 0, route_waypt_count: 0

Processing tracks
track 0 ct: 10, waypt_count: 10
track head ct: 1, track_count: 1
total track point ct: 10, track_waypt_count: 10
```