GPSBabel/gpsbabel #890: Qt6 floor work

tsteven4, 2022-07-15:
When the floor is Qt6 we can use QDateTime/QDate/QTime:fromString format arguments to utf16 string literals.

GPSBabelDeveloper, 2022-07-16:
We last released with Qt6-only builds and almost nobody noticed. I think
we've had one Windows user and one Mac user notice. I DID change the
server-side upgrade message to not bug you if you were on an OS that
Microsoft had abandoned. I can't remember if I similarly added bumpers to
Mac users. Qt 6.3 officially "only" goes back to Mojave from 2018, but I
have 6.3.1 via Homebrew on my 2012 system that's still on Catalina.

Maybe QT6.2 LTS is a reasonable place to be. Even that's almost a year old,
so it's had some air time. We're nine months into its four year cycle.

I think our 32-bit and Windows 7 days are over. Niche users are always
welcome to build from source, struggling through whatever they need, and/or
keep running old builds that they already have.

tsteven4, 2024-11-25:
This was fixed in #1278

closed 2024-11-25 by tsteven4