GPSBabel/gpsbabel #893: gpsbabel does not support gdb from basecamp 4.7.4

RalfPeter, 2022-07-18:
I get the error message that i use a wronf gdb version (less 1 or greater than 3). My basecamp is version 4.7.4 with fileversion 1.88 and programversion 4.7.
Is there no support for newer basecamp database files?

robertlipe, 2022-07-20:
I wonder if your GDB file is corrupt and essentially unrecognized.

Basecamp has been out of development for years,  I'm surprised Garmin had
any resources left on it to add a new version. Either way, Basecamp has had
the ability to save as GPX for many years and nobody else has reported
this, so even if you produced a produced a fix, we'd struggle with letting
it in.

GDB was on the list of formats I recently proposed removing. It barely hung
in there. It's a reverse-engineered format and has cost us tons of
development and maintenance effort, so I don't see putting more development
time in it. If you produce a fix (with doc, test cases, affected Basecamp
versions, etc.) and it's trivial, we'll take it but we're not going to byte
off any major development for it.
https://www.gpsbabel.org/FAQ.html#new_feature

RJL

tsteven4, 2022-07-21:
I suspect your file is just much newer than our gdb support, and not corrupted.  Does basecamp have the capability to write a downlevel gdb file?

It looks like the gdb format only supports file formats 1.7 (Ver 1), 1.8 (Ver 2), 1.9 (Ver 3).  Known later versions, mostly from basecamp, 1.15-1.88 are not supported.  It looks like the Waypoint, Route and Track record formats have all changed at some point between our support and the latest version.

basecamp was last updated May 2021.

Format clues available at https://www.memotech.franken.de/FileFormats/Garmin_MPS_GDB_and_GFI_Format.pdf, which has been updated for the latest basecamp.  The reverse engineering and versions dependencies make this format a PITA, although Herbert's work cited above certainly would make it easier to create a reader.  It could be clean slate time, perhaps with QDataStream (like qstarz_bl_1000.cc).