GPSBabel/gpsbabel #895: unmatched tags in geocache reference file

tsteven4, 2022-07-30:
1. https://github.com/GPSBabel/gpsbabel/blob/8dae11e100677217c6b4cd858a76f3bf6bc8d12c/reference/gc/GCGCA8.gpx#L46
The tags &lt;/a&gt;&lt;/p&gt; are unmatched.
This causes the output from gpsbabel -i gpx -f reference/gc/GCGCA8.gpx -o html -F GCGCA8.html to contain unmatched tags.
2. There is an additional problem.  The P end tag we add after gpsbabeldesclong is unmatched because of the UL tags in what we think is the paragraph content.

robertlipe, 2022-07-30:
>    The tags </a></p> are unmatched.

That HTML was created by hand (mine) that was rendered inside an iframe of
browsers and using crudest tools almost 20 years ago. It's even possible
that it wasn't even really written in 'real' HTML as I think the cache page
editor of that time used sort of a bbcode markup that was transcribed on
the server.

As long as we can approximately round-trip what's inside
groundspeak:long_description,
what we do with that HTML is up to us. That cache placement has long since
been removed and out of circulation. There's no need to overthink keeping
it in sync with a master copy or whatever.

We're free to change that reference file and the ones that come from it as
we see fit. We're free to edit the past if it makes the present or future
better.

Signed,
Winston Smith

tsteven4, 2023-10-25:
> 2\. There is an additional problem.  The P end tag we add after gpsbabeldesclong is unmatched because of the UL tags in what we think is the paragraph content.

Point 2. was fixed in dcdbb6a90353231497273d3062c2cd5d396e1d6e

tsteven4, 2023-10-25:
> 1. The tags ```</a></p>``` are unmatched.

Point 1. is fixed in #1196, and updated version of GCGCA8.gpx.