GPSBabel/gpsbabel #924: обмен данными с пк и эхотом

R1BGW, 2022-10-03:
Здравствуйте . Имеется эхолот GARMIN STRIKE 4 PLUS  , имеется единственный  интерфейс обмена данными  RS232 между эхолотами линейки Garmin Strike  . Имеется программа  GND GARMIN  ( уже устарела , сохраняет файлы GPX в v.1.0  не совсем корректно ) имеет возможность работать с последовательным портом и выгружать /загружать данные в эхолот в протоколе L001 .
Важной особеностью эхолота является возможность бокового сканирования и создание файлов QDC (QuickDraw Contours )
[image] сам обмен между программой и эхолотом 
[команды ASCII.txt](https://github.com/GPSBabel/gpsbabel/files/9695583/ASCII.txt)
Имеется пожелание в следующих наработках программы  GPSBabel  добавить возможность выгружать /загружать данные (треки ,точки ,глубины и т.д.)

R1BGW, 2022-10-03:
Программа GPSBabel выдает ошибку [image]
[image]

tsteven4, 2022-10-03:
Under "More Options" set the "Debugging Diagnostics" to 2 and try again.  This should dump the communication between the device and gpsbabel so you can debug it.

At some level L001 is supported.

R1BGW, 2022-10-04:
Hello. I did as you advised , the error disappeared . The GPS map file is unfortunately empty . There is no data unloading from the echo sounder .
[image]
[image]
 What's wrong? What is the error ?

tsteven4, 2022-10-04:
Unfortunately, the output to the GUI window is limited to keep it responsive.  Can you run it from the command line?  The command to run is at the top of the window in your pictures, it starts with "gpsbabel" and ends with "-F C:/Users/1/Desktop/123.gpx".

I did notice that your start and stop times from the track filter changed.  Perhaps it would be easier to debug if you just turned off the track filter to make sure it isn't throwing out any data.

R1BGW, 2022-10-05:
Hello. It was not possible to start from the command line . I get the impression that the program does not detect the device , compared to GARMIN 's DNR , when the COM port is connected , it immediately detects it .
[image]

R1BGW, 2022-10-05:
[garmin-binary.pdf](https://github.com/GPSBabel/gpsbabel/files/9714112/garmin-binary.pdf)
На данный момент нет ни одной программы  , которая могла полноценно выгружать и конвертировать из протокола L001 данные .

tsteven4, 2022-10-05:
Here is an example of L001 (Link_type 1) working, in this case with a Garmin GPS 12. This 20 year old technology still works.

>.\gpsbabel.exe -t -r -w -D 1 -i garmin -f COM3 -x validate,debug -o gpx -F garmin.gpx
GPSBabel Version: 1.8.0
main: Compiled with Qt 6.2.4 for architecture x86_64-little_endian-llp64
main: Running with Qt 6.2.4 on Windows 11 Version 2009, x86_64
main: QLocale::system() is en_US
main: QLocale() is en_US
main: QTextCodec::codecForLocale() is System, mib 0
Unit:   GPS 12 SOFTWARE  4.60
ID:     87
Version:        4.60

Capability A10:
Capability A100: D103
Capability A200: D201 D103
Capability A300: D300
Capability A400: D403
Capability A500: D501
Capability A600: D600
Capability A700: D700
Capability A800: D800
Link_type 1  Device_command 0
Waypoint: Transfer 100 Type 103
Route:    Transfer 200 Header 201 Type 103
Track:    Transfer 300 Type 300
GPS Time: Wed Dec 31 17:00:00 1969
Waypoint type: 103
Chosen waypoint length 6
options: module/option=value: validate/checkempty="0" (=default)
options: module/option=value: validate/debug="1"

Processing waypts
point ct: 453, waypt_count: 453

Processing routes
route 0 ct: 30, waypt_count: 30
route 0 ct: 29, waypt_count: 29
route 0 ct: 21, waypt_count: 21
route 0 ct: 20, waypt_count: 20
route 0 ct: 15, waypt_count: 15
route 0 ct: 8, waypt_count: 8
route 0 ct: 4, waypt_count: 4
route 0 ct: 18, waypt_count: 18
route 0 ct: 30, waypt_count: 30
route 0 ct: 18, waypt_count: 18
route 0 ct: 3, waypt_count: 3
route 0 ct: 30, waypt_count: 30
route 0 ct: 16, waypt_count: 16
route 0 ct: 8, waypt_count: 8
route 0 ct: 22, waypt_count: 22
route 0 ct: 14, waypt_count: 14
route 0 ct: 10, waypt_count: 10
route 0 ct: 18, waypt_count: 18
route 0 ct: 29, waypt_count: 29
route 0 ct: 17, waypt_count: 17
route head ct: 20, route_count: 20
total route point ct: 360, route_waypt_count: 360

Processing tracks
track 0 ct: 1001, waypt_count: 1001
track head ct: 1, track_count: 1
total track point ct: 1001, track_waypt_count: 1001
options: module/option=value: gpx/snlen="32" (=default)
options: module/option=value: gpx/elevprec="3" (=default)

tsteven4, 2022-10-05:
Here is what happens if the INTERFACE mode on the GPS 12 is NONE/NONE instead of GRMN/GRMN like it was above.

>.\gpsbabel.exe -t -r -w -D 1 -i garmin -f COM3 -x validate,debug
GPSBabel Version: 1.8.0
main: Compiled with Qt 6.2.4 for architecture x86_64-little_endian-llp64
main: Running with Qt 6.2.4 on Windows 11 Version 2009, x86_64
main: QLocale::system() is en_US
main: QLocale() is en_US
main: QTextCodec::codecForLocale() is System, mib 0
[ERROR] GPS_Packet_Read: Timeout.  No data received.
GARMIN:Can't init COM3

tsteven4, 2022-10-05:
and here is what happens if the INTERFACE mode isNMEA/NMEA.  **Note that this looks like what you originally reported.**  Also, the error message makes some sense, the device is sending NMEA packets instead of using the garmin serial protocol.

>.\gpsbabel.exe -t -r -w -D 1 -i garmin -f COM3 -x validate,debug
GPSBabel Version: 1.8.0
main: Compiled with Qt 6.2.4 for architecture x86_64-little_endian-llp64
main: Running with Qt 6.2.4 on Windows 11 Version 2009, x86_64
main: QLocale::system() is en_US
main: QLocale() is en_US
main: QTextCodec::codecForLocale() is System, mib 0
GPS_Packet_Read: No DLE.  Data received, but probably not a garmin packet.
GARMIN:Can't init COM3

R1BGW, 2022-10-06:
After dancing with a tambourine , something began to turn out . By RX error .

C:\Program Files\GPSBabel>gpsbabel.exe -t -r -w -D 2 -i garmin,baud=115200 -f COM5 -x \dev\ttyUSB0 -o gpx -F garmin.gpx

GPSBabel Version: 1.8.0
main: Repository SHA: 3054af94a308f2fafdac7e2f2862ea0f6b335666
main: Compiled with Qt 6.2.2 for architecture x86_64-little_endian-llp64
main: Running with Qt 6.2.2 on Windows 6.3, x86_64
main: QLocale::system() is ru_RU
main: QLocale() is ru_RU
main: QTextCodec::codecForLocale() is System, mib 0
options: module/option=value: garmin/baud="115200"
Opening COM5
Tx Data:10 fe 00 02 10 03 : ...(PRDREQ  )
Rx Data:[ERROR] GPS_Packet_Read: Timeout.  No data received.
GARMIN:Can't init COM5

R1BGW, 2022-10-07:
Доброго времени суток . Танец с бубном , это от северных народов , шаманство . Это устройство посылает посылку  1 раз в 10 секунд на последовательный порт RS232 .
[image]
Я исправил в командной строке  
C:\Program Files\GPSBabel>gpsbabel.exe -t -r -w -D 2 -i garmin,baud=115200 -f
GPSBabel Version: 1.8.0
main: Repository SHA: 3054af94a308f2fafdac7e2f2862ea0f6b335666
main: Compiled with Qt 6.2.2 for architecture x86_64-little_endian-llp64
main: Running with Qt 6.2.2 on Windows 6.3, x86_64
main: QLocale::system() is ru_RU
main: QLocale() is ru_RU
main: QTextCodec::codecForLocale() is System, mib 0
options: module/option=value: garmin/baud="115200"
No file or device name specified.

GPSBabelDeveloper, 2022-10-07:
No file or device name specified.

You left off the "com5:"

R1BGW, 2022-10-07:
Microsoft Windows [Version 6.3.9600]
(c) Корпорация Майкрософт (Microsoft Corporation), 2013. Все права защищены.

C:\Users\1>cd C:\Program files\gpsbabel

C:\Program Files\GPSBabel>gpsbabel.exe -t -r -w -D 2 -i garmin,baud=115200 -fCOM
5 -x
GPSBabel Version: 1.8.0
main: Repository SHA: 3054af94a308f2fafdac7e2f2862ea0f6b335666
main: Compiled with Qt 6.2.2 for architecture x86_64-little_endian-llp64
main: Running with Qt 6.2.2 on Windows 6.3, x86_64
main: QLocale::system() is ru_RU
main: QLocale() is ru_RU
main: QTextCodec::codecForLocale() is System, mib 0
options: module/option=value: garmin/baud="115200"
Opening COM5
Tx Data:10 fe 00 02 10 03 : ...(PRDREQ  )
Rx Data:[ERROR] GPS_Packet_Read: Timeout.  No data received.
GARMIN:Can't init COM5

tsteven4, 2022-10-07:
Your serial trace appears to be similar to https://blog.openwww.org/garmin-high-speed-protocol-or-network/
The striker is suppose to use something called Garmin "High Speed Protocol".  If we try to interpret the data from blog.openwww.org using

> Garmin Device Interface Specification 
> September 16, 2004
> Drawing Number: 001-00063-00 Rev. B

some issues come up.

byte |   | data(hex) | data(dec) |   |   |  
-- | -- | -- | -- | -- | -- | --
0 |   | 10 | 16 | DLE |   |  
1 |   | 7B | 123 |   | Packet ID identifies the type of packet |  
2 |   | 12 | 18 |   | Size of Packet Data number of bytes of packet data (bytes 3 to n-4) |  
3 | 1 | 0C | 12 |   |   |  
4 | 2 | 0 | 0 |   |   |  
5 | 3 | 0 | 0 |   |   |  
6 | 4 | 0 | 0 |   |   |  
7 | 5 | FD | 253 |   |   |  
8 | 6 | 9 | 9 |   |   |  
9 | 7 | 17 | 23 |   |   |  
10 | 8 | AA | 170 |   |   |  
11 | 9 | DF | 223 |   |   |  
12 | 10 | E9 | 233 |   |   |  
13 | 11 | 9 | 9 |   |   |  
14 | 12 | 0 | 0 |   |   |  
15 | 13 | 9 | 9 |   |   |  
16 | 14 | 0 | 0 |   |   |  
17 | 15 | 2 | 2 |   |   |  
18 | 16 | 0 | 0 |   |   |  
19 | 17 | 0 | 0 |   |   |  
20 | 18 | 0 | 0 |   |   |  
21 | 19 | C4 | 196 |   |   |  
22 | 20 | 10 | 16 |   |   |  
23 | 21 | 3 | 3 |   |   |  
24 | 22 | AA | 170 |   |   |  
25 | 23 | DF | 223 |   |   |  
26 | 24 | E9 | 233 |   |   |  
27 | 25 | 9 | 9 |   |   |  
28 | 26 | 0 | 0 |   |   |  
29 | 27 | 9 | 9 |   |   |  
30 | 28 | 0 | 0 |   |   |  
31 | 29 | 2 | 2 |   |   |  
32 | 30 | 0 | 0 |   |   |  
33 | 31 | 0 | 0 |   |   |  
34 | 32 | 0 | 0 |   |   |  
35 |   | C4 | 196 |   | Checksum 2's complement of the sum of all bytes from byte 1 to byte n-4 |  
36 |   | 10 | 16 | DLE |   |  
37 |   | 3 | 3 | ETX |   |  

- packet id 123. is an unknown packet type
- the size of the packet does not match the expected number of bytes for a 38 byte packet
- the checksum doesn't match

I haven't found any specification for Garmin High Speed Protocol, or anybody claiming they know how to decode it.  The support in the gpsbabel garmin format follows the "Garmin Device Interface Specification" cited above, which was written long before the Garmin Striker series.

R1BGW, 2022-10-08:
Здравствуйте. Хочу поблагодарить за оказанную помощь и оказанную поддержку . Я буду продолжать искать решение . Спасибо и удачи .

R1BGW, 2022-11-05:
> 

Hello. I have looked through your blog , and it would be interesting to be aware of your developments on Arduino . Unfortunately, I am not a programmer, but by profession I am connected with electronics and radio engineering. I have some small developments ( so far only in theory ) of creating a buffer on the Arduino via the RS 232 interface and outputting information from the device to the SD card . I tried to figure out the GARMIN encoding , not quite everything is clear . I will continue to study and develop .
[garmin-binary.pdf](https://github.com/GPSBabel/gpsbabel/files/9943477/garmin-binary.pdf)

R1BGW, 2022-11-05:
> В субботу, 5 ноября 2022 года, в 4:36 утра R1BGW ***@***.***> написал: Здравствуйте. Я просмотрел ваш блог, и было бы интересно узнать о ваших разработках в Arduino .
> GPSBabel слишком требователен к ATmega / Arduino. Действительно, последовательный протокол Garmin сложнее, чем я бы попробовал с Arduino, но если вы создаете с нуля, это возможно.
> буферизация на Arduino через интерфейс RS 232 и вывод информации с устройства на SD-карту . Я попытался разобраться с кодировкой GARMIN, не совсем все понятно . Я буду продолжать учиться и развиваться . garmin-двоичный файл.pdf <https://github.com/GPSBabel/gpsbabel/files/9943477/garmin-binary.pdf >
> Большая часть * этой * спецификации реализована в GPSBabel. См.: https://github.com/GPSBabel/gpsbabel/tree/master/jeeps Но эта спецификация не является Нападающей. Этот код будет слишком тяжелым для Arduino. Код, подобный https://github.com/GPSBabel/gpsbabel/blob/master/jeeps/gpsserial.cc возможно, у вас есть подсказки, которые вы можете использовать. Но мы делаем такие вещи, как выделение и освобождение пакетов, которые плохо подходят для 8-битного ядра с 2 КБ оперативной памяти. Этот код представляет собой смесь C и C ++, и совсем не так, как я бы написал его с нуля сегодня, но он проверен в бою. ОДНАКО этот код поддерживает сотни Garmins, но не Striker. "Высокоскоростной протокол", который вы описали ранее, не соответствует этой спецификации. Наш код может вас отвлечь. 1) Это не тот протокол устройства, который вам нужен. 2) Он был создан для 32-разрядных процессоров размером с настольный компьютер, где копирование и распределение просты, и мы быстрее, чем GPS, со многими буферами под нами. Ни один из них не соответствует действительности в ATmega. Удачи. RJL PS Я тоже делаю микроконтроллеры, но больше, чем arduino. [https://www.robertlipe.com /](https://www.robertlipe.com/)
> […](#)
> Идентификатор сообщения: ***@***.***>

I tried to connect to the device in GPSBabel on the command line , but failed every time . An error occurred on initialization of the COM port or timeout . The program would like to receive an answer in the form of information about the device. Perhaps I did not correctly register the commands . If this feature is implemented in the program, according to the specification, I would like to see the spelling options on the command line ( at least the request and response of the device ).

GPSBabelDeveloper, 2022-11-05:
We did this a month ago, didn't we?

Your GPS uses a protocol we don't know about. There is no mistake on your
part. No spelling options on the command line will help. If you wish to
implement your own, using our code for guidance, please feel free.

From what you showed us (Steven) before, there is no support for your
device in our code.