GPSBabel/gpsbabel #943: Track filter 'move' needlessly requires ordered timestamps

aklomp, 2022-11-06:
I use `gpsbabel` to download tracks from my QStarz BT-Q1000XT. Because of gps week rollover, I need to add 1024 weeks to the output. I do this with the `move` track filter, using the following snippet:

```sh
# Download track to a temporary file, add 1024 weeks.
gpsbabel -t -w \
  -D 2 \
  -i mtk,erase,block_size_kb=64 \
  -f /dev/ttyACM0 \
  -x track,move=$((1024 * 7))d \
  -o gpx \
  -F out.gpx
```

Now, for some reason, my GPS occasionally stores trackpoints in non-chronological order. I'm not sure why; it could have something to do with storing estimated positions with an approximated time when no GPS fix is available, or something like that. The point is, it happens.

However, when this happens, the snippet above no longer works because the `move` filter complains about unordered timestamps and exits the program. In order to silence the error, I need to `merge` all tracks together, which does so at the cost of squishing all separate tracks together and creating a mess.

I dove into the code and it looks like the `move` filter does not intrinsically need timestamps to be ordered at all. It operates on each timestamp in isolation and has no reason to require ordered timestamps.

 The reason it's triggering the error is because in [`trackfilter.cc`](https://github.com/GPSBabel/gpsbabel/blob/master/trackfilter.cc), `opt_move` is [made part of the `need_time` group](https://github.com/GPSBabel/gpsbabel/blob/master/trackfilter.cc#L962), and all members of that group trigger a check of whether timestamps are ordered.

This is overly restrictive for `move`. I think the solution would be to create a new group that contains all filters that need *ordered* time and only run the order check for those filters. The `need_time` filter would be constructed of that group plus `move` and whatever other filters do not need ordered timestamps.

By the way, I patched this locally by removing `opt_move` from the `need_time` group and verifying that the error is no longer triggered. I use that patch as a temporary fix for my own setup.

tsteven4, 2022-11-11:
BTW, Qstarz has updated firmware that may solve the gps week rollover problem for you, see http://www.qstarz.com/news/news-20190408.html and https://qws.qstarz.com/WNRO/

aklomp, 2022-11-11:
Indeed, I noticed that shortly after posting this issue :)  I haven't applied the update yet. Which brings me to another point - I was planning on doing a full capture of the USB traffic between the firmware uploader and my device, by running the updater inside a VM with USB passthrough and logging enabled. Would that traffic dump be of any interest to you guys? It would contain the complete NMEA conversation needed to write firmware to the device.

Then again, if you download Qstarz' update tool and look inside the binary, you can already see a bunch of NMEA commands.

tsteven4, 2022-11-11:
> -x track,move=$((1024 * 7))d \
we added support for "move=1024w" in 1.7.0 and 1.8.0

I don't see a need for gpsbabel to update Qstarz fimware, that sounds like something better left to Qstarz.  Thanks anyway.

PR #945 (2022-11-12): relax time constraints on trackfilter move.
For track filter move operations:
1. don't require all track points to have time, just leave timeless track points alone.
2. don't require track points to be ordered by time for move.

This also results in the tracks not being sorted by time for move only operations.  Instead the original order is preserved.

This resolves #943
closed 2022-11-12 by tsteven4