GPSBabel/gpsbabel #955: kml option? for html_encrypt never implemented

tsteven4, 2022-12-06:
html_encrypt was introduced in https://github.com/GPSBabel/gpsbabel/commit/3266f6df7, but there isn't a way to set it.  Perhaps it was suppose to be an option?

tsteven4, 2022-12-06:
I'll drop html_encrypt from kml.cc entirely, i.e. assume html_encrypt = false.

I note that the html format has a html_encrypt option that appears to be fully implemented.

GPSBabelDeveloper, 2022-12-07:
I think it's a different use case. People using the HTML mode are probably
making a "book" of caches that they're finding in a day and may legit want
to read the read the description and logs and such while they're hunting
and not necessarily see a spoiler.

My original plan was to mimic what the site itself did (under my guidance)
years ago and mark up a div with the hint that was rot13 encoded and add a
button to trigger the rot13 rewrite of that div. I did this when Javascript
in KML balloons was still pretty new (it was a test case, in fact..) and
clearly never finished it.

I think that Javascript on Earth for Mobile (at least for Android) is
explicitly disabled for security reasons. Hunting geocaches with Earth
while on mobile would be incredibly awkward and I'm pretty sure that people
just don't do that. People DO use Geocaching GPX -> KML on desktop Earth to
visualize caches in an area on a map and occasionally pop a balloon.
Dragging and dropping a Pocket Query (a GPX that contains all that
"groundspeak:" stuff) into Earth (Earth uses an internal GPSBabel to handle
the import) is a fairly delightful thing when people discover that it works.

As nobody's noticed that the option isn't actually available in 12 years, I
dont feel bad about just dropping the unreachable code.

I'll whack it, but first  have to go debug the build itself.  Cmake is
fussing that it can't
find /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk
on my system that has a
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.sdk
so something somewhere has a cached value of the SDK location. This system
is still on Monterey (12.6.1) which is last year's release, so it's not
like it's even on the 2021 MacOS. I'll go figure that out now...

I can't say that debugging a failing cmake build is that much more fun than
debugging a failed configure/make cycle. Maybe it is and I just have 30
years less experience at it. :-)