GPSBabel/gpsbabel #972: upgrade check failure

tsteven4, 2023-01-04:
Ubuntu finally released the fix to Qt6 QLibraryInfo for jammy.  But we have a couple of new issues:
1) If we download from Qt then the expected OpenSSL versions aren't compatible.  jammy has 3.0.2, Qt expects 1.

We get the popup
[image]
 and the console shows:

```
qt.tlsbackend.ossl: Incompatible version of OpenSSL (built with OpenSSL 1.x, runtime version is >= 3.x)
qt.network.ssl: The backend "cert-only" does not support QSslKey
qt.network.ssl: Active TLS backend does not support key creation
qt.network.ssl: The backend "cert-only" does not support QSslKey
qt.network.ssl: Active TLS backend does not support key creation
qt.network.ssl: The backend "cert-only" does not support QSslKey
qt.network.ssl: Active TLS backend does not support key creation
qt.network.ssl: The backend "cert-only" does not support QSslKey
qt.network.ssl: Active TLS backend does not support key creation
qt.network.ssl: The backend "cert-only" does not support QSslKey
qt.network.ssl: Active TLS backend does not support key creation
qt.network.ssl: The backend "cert-only" does not support QSslKey
qt.network.ssl: Active TLS backend does not support key creation
qt.network.ssl: The backend "cert-only" does not support QSslSocket
qt.network.ssl: The backend named "cert-only" does not support TLS
qt.network.ssl: QSslSocket::connectToHostEncrypted: TLS initialization failed
```

2) If we use the ubuntu supplied Qt6, then (testing true, qDebug() << oresponse)
(also notice two periods in the popup)

[image]

```
Warning: Ignoring WAYLAND_DISPLAY on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
http status code  200
"NULL\nNULL\n0"
```

Now, if we change the upgrade check method from http to https then the testing message notifying the user an upgrade from 1.3.1 is presented as expected.

tsteven4, 2023-01-04:
and Qt 6.2.4

tsteven4, 2023-01-04:
Also see https://www.qt.io/blog/moving-to-openssl-3-in-binary-builds-starting-from-qt-6.5-beta-2

tsteven4, 2023-01-09:
The issue with the popup "Invalid return data at line 1: Start tag expected.." also occurs on macOS.  It appears that gpsbabel.org doesn't respond well to redirects which are reissued using GET.  upgrade issues a POST using the http scheme.

> Note: When Qt handles redirects it will, for legacy and compatibility reasons, issue the redirected request using GET when the server returns a 301 or 302 response, regardless of the original method used, unless it was HEAD.

 @robertlipe can you please look at this?

tsteven4, 2023-06-20:
The invalid return data popup has been reported again on [gpsbabel-misc](https://sourceforge.net/p/gpsbabel/mailman/message/37859810/)

tsteven4, 2023-06-20:
The invalid return data popup is reproducible with 043fe4f0084a0abe2dfe095f3a23725de87590df on
ProductName:	macOS
ProductVersion:	12.6.4

GPSBabelDeveloper, 2023-06-21:
Argh. Another silent server-side change. "Thanks, Dreamhost/$LANGUAGE".

I've modified it to allow the upgrade query to present upgrades. Stats
collection will resume in the next couple of days.

curl https://www.gpsbabel.org/upgrade_check.html should again return joy.
Windows 7 and 8 and other "we have no upgrade for you" cases should work as
expected.

Sorry this is so messy. Dreamhost rolls the upgrade check out, verifies
that our home page renders, and considers it done. upgrade_check is a very
different beast internally. Maybe I have to start running a nightly crafted
check to do.a form of automated CI on the server-side.

But for now, the user-visible parts should again work.

RJL