GPSBabel/gpsbabel #981: "All track points have been dropped!" but end up in file anyway

jidanni, 2023-01-12:
```
gpsbabel -t -i kml -f /tmp/k.kml -x track,start=20221220204931,stop=20221220204931 -o csv -F /tmp/l.csv
trackfilter-range: All 36755 track points have been dropped!
wc /tmp/l.csv
  36757  110271 9449368 /tmp/l.csv
```
If I didn't use -F then they really would have been dropped. I am just following the example on
https://www.gpsbabel.org/htmldoc-1.8.0/filter_track.html#ex_track_startstop

jidanni, 2023-01-12:
I see. A previous command had turned them into <placemark>s. Closing.

closed 2023-01-12 by jidanni

jidanni, 2023-01-12:
OK, it calls waypoints "trackpoints" perhaps.
Anyway no combination of -t or -w or -x track,start=1,stop=2023111111 will get rid of that message,
"trackfilter-range: All 36755 track points have been dropped!"

jidanni, 2023-01-12:
[Makefile.txt](https://github.com/GPSBabel/gpsbabel/files/10398307/Makefile.txt)
[FlightAware_JAL57_KSFO_RJAA_20221220.kml.gz](https://github.com/GPSBabel/gpsbabel/files/10398310/FlightAware_JAL57_KSFO_RJAA_20221220.kml.gz)
I was just trying to extract one point (one second) from the KML.

tsteven4, 2023-01-12:
What version of gpsbabel are you using? This works for me.  If you want to get rid of the waypoints leaving just the one trackpoint use the nuke filter.

./bld/gpsbabel -i kml -f FlightAware_JAL57_KSFO_RJAA_20221220.kml  -x track,start=20221220204931,stop=20221220204931 -x validate,debug -o kml,units=a,extrude=1 -F Flight.kml

Processing waypts
point ct: 2, waypt_count: 2

Processing routes
route head ct: 0, route_count: 0
total route point ct: 0, route_waypt_count: 0

Processing tracks
track 0 ct: 1, waypt_count: 1
track head ct: 1, track_count: 1
total track point ct: 1, track_waypt_count: 1

jidanni, 2023-01-17:
I also notice that "if we put the clothes back in the washing machine a second time" things change. But not a third time:
```
gpsbabel -i kml -f FlightAware_JAL57_KSFO_RJAA_20221220.kml \
-x track,start=2022122020,stop=2022122021 -o kml,units=a,extrude=1 -F - | tee /tmp/z.kml | wc -l
865
#
gpsbabel -i kml -f /tmp/z.kml \
-x track,start=2022122020,stop=2022122021 -o kml,units=a,extrude=1 -F - | tee /tmp/zz.kml | wc -l
trackfilter-range: All 26 track points have been dropped!
557
#
gpsbabel -i kml -f /tmp/zz.kml \
-x track,start=2022122020,stop=2022122021 -o kml,units=a,extrude=1 -F - | tee /tmp/zzz.kml | wc -l
557

$ grep -i styleurl z.kml zz.kml|perl -pwle 's/\s+//;'|sort|uniq -c|sort -rn|head -n 2
     28 zz.kml:<styleUrl>#waypoint</styleUrl>
     26 z.kml:<styleUrl>#track</styleUrl>
```

jidanni, 2023-01-17:
And I just want to get the one point here, but I get lots,
```
gpsbabel -i kml -f /tmp/zz.kml \
-x track,start=20221220205714,stop=20221220205714 -o kml -F - | grep when | head -n 5
          <when>2022-12-20T20:49:31Z</when>
          <when>2022-12-20T20:49:47Z</when>
          <when>2022-12-20T20:50:03Z</when>
          <when>2022-12-20T20:50:19Z</when>
          <when>2022-12-20T20:50:35Z</when>
```
and if I use -x nuketypes,... I get none.

GPSBabelDeveloper, 2023-01-17:
I don't see the KML attached to this report for repro, but IIRC you were on
some ancient version of GPSBabel.

Before Jan 6, 2019, all the trackfilter matching happened on second
boundaries. You have four points that are within the same second. (And one
that's not, which I can't readily explain, but I'm not going to guess.)

https://github.com/GPSBabel/gpsbabel/commit/3816808308bb78b41538bd66cdad925502270fba#diff-d9a5902bea6a61f6ceda7ee1d864030599a8e8227627392983024bff959bd341

tsteven4, 2023-01-17:
> I also notice that "if we put the clothes back in the washing machine a second time" things change. (But no changes the third time):
> 
> ```
> gpsbabel -i kml -f FlightAware_JAL57_KSFO_RJAA_20221220.kml \
> -x track,start=2022122020,stop=2022122021 -o kml,units=a,extrude=1 -F - | tee /tmp/z.kml | wc -l
> 865
> #
> gpsbabel -i kml -f /tmp/z.kml \
> -x track,start=2022122020,stop=2022122021 -o kml,units=a,extrude=1 -F - | tee /tmp/zz.kml | wc -l
> trackfilter-range: All 26 track points have been dropped!
> 557
> #
> gpsbabel -i kml -f /tmp/zz.kml \
> -x track,start=2022122020,stop=2022122021 -o kml,units=a,extrude=1 -F - | tee /tmp/zzz.kml | wc -l
> 557
> 
> $ grep -i styleurl z.kml zz.kml|perl -pwle 's/\s+//;'|sort|uniq -c|sort -rn|head -n 2
>      28 zz.kml:<styleUrl>#waypoint</styleUrl>
>      26 z.kml:<styleUrl>#track</styleUrl>
> ```

The kml reader creates tracks from a variety of kml elements.  One is /Placemark/LineString/coordinates.  Another is /Placemark/*gx:Track.  LineStrings do not have time data.  gx:Tracks may have time data.

The kml writer will write LineStrings if the lines options is on, which it is by default.
The kml writer will write gx:Tracks if the track options is on, which is off by default.
So if you want to preserve times with track points in kml, and not create one version of the track without times and one with times then you should use -o kml,track=1,lines=0.

To see what the trackfilter is doing using the validate filter with the debug option before and after running the track filter.  The trackfilter only operates on tracks, not waypoints or routes.  This will print out counts for waypoints, routes and tracks as shown below:
```
+ ./bld/gpsbabel -i kml -f FlightAware_JAL57_KSFO_RJAA_20221220.kml -x validate,debug -x track,start=2022122020,stop=2022122021 -x validate,debug -o kml,units=a,extrude=1,track=1,lines=0 -F /tmp/z.kml

Processing waypts
point ct: 2, waypt_count: 2

Processing routes
route head ct: 0, route_count: 0
total route point ct: 0, route_waypt_count: 0

Processing tracks
track 0 ct: 458, waypt_count: 458
track head ct: 1, track_count: 1
total track point ct: 458, track_waypt_count: 458

Processing waypts
point ct: 2, waypt_count: 2

Processing routes
route head ct: 0, route_count: 0
total route point ct: 0, route_waypt_count: 0

Processing tracks
track 0 ct: 26, waypt_count: 26
track head ct: 1, track_count: 1
total track point ct: 26, track_waypt_count: 26
+ ./bld/gpsbabel -i kml -f /tmp/z.kml -x validate,debug -x track,start=2022122020,stop=2022122021 -x validate,debug -o kml,units=a,extrude=1,track=1,lines=0 -F /tmp/zz.kml

Processing waypts
point ct: 28, waypt_count: 28

Processing routes
route head ct: 0, route_count: 0
total route point ct: 0, route_waypt_count: 0

Processing tracks
track 0 ct: 26, waypt_count: 26
track head ct: 1, track_count: 1
total track point ct: 26, track_waypt_count: 26

Processing waypts
point ct: 28, waypt_count: 28

Processing routes
route head ct: 0, route_count: 0
total route point ct: 0, route_waypt_count: 0

Processing tracks
track 0 ct: 26, waypt_count: 26
track head ct: 1, track_count: 1
total track point ct: 26, track_waypt_count: 26
+ ./bld/gpsbabel -i kml -f /tmp/zz.kml -x validate,debug -x track,start=2022122020,stop=2022122021 -x validate,debug -o kml,units=a,extrude=1,track=1,lines=0 -F /tmp/zzz.kml

Processing waypts
point ct: 54, waypt_count: 54

Processing routes
route head ct: 0, route_count: 0
total route point ct: 0, route_waypt_count: 0

Processing tracks
track 0 ct: 26, waypt_count: 26
track head ct: 1, track_count: 1
total track point ct: 26, track_waypt_count: 26

Processing waypts
point ct: 54, waypt_count: 54

Processing routes
route head ct: 0, route_count: 0
total route point ct: 0, route_waypt_count: 0

Processing tracks
track 0 ct: 26, waypt_count: 26
track head ct: 1, track_count: 1
total track point ct: 26, track_waypt_count: 26
```

tsteven4, 2023-01-17:
Furthermore, if you don't want to create waypoints from the tracks you should use the points=0 options when writing kml.  In the test above I didn't use it.  You can see you started with two
```
Processing waypts
point ct: 2, waypt_count: 2
```
but the second time you read 28:
```
Processing waypts
point ct: 28, waypt_count: 28

```

tsteven4, 2023-01-17:
> And I just want to get the one point here, but I get lots,
> 
> ```
> gpsbabel -i kml -f /tmp/zz.kml \
> -x track,start=20221220205714,stop=20221220205714 -o kml -F - | grep when | head -n 5
>           <when>2022-12-20T20:49:31Z</when>
>           <when>2022-12-20T20:49:47Z</when>
>           <when>2022-12-20T20:50:03Z</when>
>           <when>2022-12-20T20:50:19Z</when>
>           <when>2022-12-20T20:50:35Z</when>
> ```
> 
> and if I add -x nuketypes,... I get none.

No.  You actually didn't start or end with any trackpoints (track_waypt_count = 0).  You did have waypoints(waypt_count) which the track filter will not alter.  Two of the waypoints where from the original Flightaware kml, and 26 were created when you wrote you first kml file without using the points=0 option.
```
gpsbabel -i kml -f /tmp/zz.kml -x validate,debug -x track,start=20221220205714,stop=20221220205714 -x validate,debug -o kml -F /dev/null

Processing waypts
point ct: 28, waypt_count: 28

Processing routes
route head ct: 0, route_count: 0
total route point ct: 0, route_waypt_count: 0

Processing tracks
track head ct: 0, track_count: 0
total track point ct: 0, track_waypt_count: 0

Processing waypts
point ct: 28, waypt_count: 28

Processing routes
route head ct: 0, route_count: 0
total route point ct: 0, route_waypt_count: 0

Processing tracks
track head ct: 0, track_count: 0
total track point ct: 0, track_waypt_count: 0
```

jidanni, 2023-01-17:
I was using current https://packages.debian.org/sid/gpsbabel but can you believe they don't have a single date on the entire website.

https://github.com/GPSBabel/gpsbabel/issues/981#issuecomment-1379812632 has the source KML.
Wow sure wasn't paying attention to all those features you guys mentioned. No wonder I had problems. OK.