GPSBabel/gpsbabel #982: garmin, ozi, xcsv mkshort options whitespace_ok, must_unique broken since 2006

tsteven4, 2023-01-12:
I think https://github.com/GPSBabel/gpsbabel/commit/75a4b039b broke these options by adding the logic in vecs noted by
`/* for bool options without default: don't set argval if "FALSE" */`

at that time, as today, mkshort_new_handle set both options to 1 (true).

garmin.c, ozi.c and xcsv.c wouldn't call setshort_whitespace_ok or setshort_mustuniq if the option value was null.  They all used bool options without a default.  Therefore, it was impossible to use those options to set whitespace_ok to false or must_unique to false.

If so this went undiscovered for 16 years.  This suggest to me these options can be removed from these formats instead of fixed.

The option must_upper defaulted to 0 in mkshort_new_handle.  Therefore the format options to set this could have worked.  I have verified that they don't work today with ozi and xcsv.  I haven't verified that the above commit broke them, or that they stayed broken for every release since then.

I would propose removing the format options for whitespaceok and must_unique from garmin, ozi and xcsv.  We could further consider removing the option for must_upper, although that worked and could be in use.

tsteven4, 2023-01-12:
I was able to rebuild 1.6.0 (2019) and verify the non-functionality of those two options with ozi and xcsv.

tsteven4, 2023-01-12:
I was also able to build 1.5.4 (2017), with a few tweaks, and verify the same non-functionality.

GPSBabelDeveloper, 2023-01-12:
Agreed. After even five years,. It's not a bug; it's a feeeeeature.

tsteven4, 2023-01-12:
@robertlipe 
Specifically I propose to delete
ozi snwhite, snunique and snupper*
garmin snwhite
xcsv snwhite, snunique and snupper*
options.

The snupper* options did work, so the lack of a need for them is unproven.  I contend the others have been ignored worked for 16 years.

tsteven4, 2023-11-20:
I'm ok with the idea, but #1239 has a bunch of merge conflicts.

tsteven4, 2023-11-20:
#1239 looks like a bad merge.  Sorry, I have touched a ton of stuff in the last week converting the last formats to Format class, including garmin.

PR #1351 (2024-10-18): refactor option handling
This simplifies option usage and related memory management.  It also resolves #982.  I originally anticipated additional derived classes of Option for other argument types, but this is quite a large changeset already.
commit 8466bfbd (2024-10-18): refactor option handling (#1351)

* rough in Option class

* testo passes

* rough in OptionBool

* more OptionBool use

* more OptionBool

* fix isEmpty

* store qstrings

* dont return const

* rename

* convert a few lost ARGTYPE_BOOLs to OptionBool.

* utilize avalable QString for options.

* obsolete argvalptr

* Option is abstract class

* eliminate special handling of ARGTYPE_BOOL without default in vecs.

* fix #982 ...

garmin, ozi, xcsv mkshort options whitespace_ok, must_unique broken since 2006.

* test to make sure OptionBool iff ARGTYPE_BOOL

* use OptionBool for all ARGTYPE_BOOL [... cut]
closed 2024-10-18 by tsteven4