GPSBabel/gpsbabel #991: Mention interpolation not a great circle

jidanni, 2023-01-20:
The GPSbabel interpolation documentation perhaps should mention that yes  such straight line interpolation might not be  best for airplane routes, which are often great circles.

https://github.com/dfandrich/gpscorrelate/pull/21

tsteven4, 2023-01-20:
The interpolation filter does creates points along a great circle, but planes may not go along it.
[image]
The image shows an interpolated track between Tokyo and San Francisco along with your FlightAware track.

jidanni, 2023-01-22:
I see. Okay, then it is even more important to mention that the interpolations are great circles, because one would  expect straight lines.