GPSBabel/gpsbabel #1002: Missing version number in GPSBabel 1.8.0 (MacOS)

suschizu, 2023-02-08:
hello - the latest download doesn't contain a version number -  the bundle version and short version string key inside the info.plist
are missing completely:

the info.plist should contain:
```
	<key>CFBundleShortVersionString</key>
	<string>1.8.0</string>
```
and 
	```
<key>CFBundleVersion</key>
	<string>1234</string>
```

DL: https://www.gpsbabel.org/plan9.php

<img width="356" alt="Screenshot 2023-02-08 at 19 56 43" src="https://user-images.githubusercontent.com/16004333/217625691-e2efcf44-41a9-4d82-b8fc-4d8ce4c1bdb5.png">

tsteven4, 2023-02-08:
This will be fixed in the next release.  The versions haven't been rolled for the next release yet, but the current build system produces an Info.plist like the following.  This is related to the move from qmake to cmake.
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple Computer//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>CFBundleInfoDictionaryVersion</key>
        <string>6.0</string>
        <key>CFBundlePackageType</key>
        <string>APPL</string>

        <key>CFBundleName</key>
        <string>GPSBabel</string>
        <key>CFBundleIdentifier</key>
        <string>org.gpsbabel.gpsbabelfe</string>
        <key>CFBundleExecutable</key>
        <string>GPSBabelFE</string>

        <key>CFBundleVersion</key>
        <string>32</string>
        <key>CFBundleShortVersionString</key>
        <string>1.8.0</string>

        <key>LSMinimumSystemVersion</key>
        <string></string>

        <key>NSHumanReadableCopyright</key>
        <string>© 2002-2023 Robert Lipe</string>

        <key>CFBundleIconFile</key>
        <string>appicon.icns</string>

        <key>CFBundleDevelopmentRegion</key>
        <string>English</string>

        <key>NSPrincipalClass</key>
        <string>NSApplication</string>

        <key>NSSupportsAutomaticGraphicsSwitching</key>
        <true/>
</dict>
</plist>
```

suschizu, 2023-02-08:
great thanks!

closed 2023-02-08 by suschizu